littlefs-project/littlefs #796: Content overwriting issue in a text file generated by LittleFS using STM32F401RE.

smitpatel30, 2023-04-12:
Hi, Developers..

I have created a text file using LittleFS file system in W25q16JV SPI Flash Memory. I want to store the time stamps of RTC in a text file. Whenever I store a new data or time stamp  in a text file, the new data overwrites the previous one that had been stored in a file. At the last, I left with only one timestamp. So, the process is like.

1.Calling RTC
2. Latching Data
3.Mount the file system
4. Open the file as write only
5. Writing first Timestamp
6. Closing the file
7. Unmounting the file 
These steps repeat after each iteration. But , after each iteration the current latched data replaced the previous one. 

On the other side, If I use a different strategy like below , then It works perfectly. It stores the both timestamps.. Also, I am able to read both the timestamps in a buffer.

1.Calling RTC
2. Latching Data
3.Mount the file system
4. Open the file as write only
5. Writing a random dummy timestamp-1
6.  Writing a random dummy timestamp-2
7. Closing the file
8. Unmounting the file

alexandrehagihara, 2023-04-12:
If I understood correctly, you want to append a new timestamp to a file. To do this, between steps 4 and 5 you should call lfs_file_seek(lfs,file,0,LFS_SEEK_END).

smitpatel30, 2023-04-17:
Thanks @alexandrehagihara for helping me out ... By  your Advise, I had successfully created a file using LittleFS and stored many lines of data in a text file. However, I now want to transfer this file system data from W25QXX SPI Flash memory to my Windows10 Laptop. Is it possible to read a file by using a specific driver in windows ,that read data from W25QXX SPI Flash memory?... If it is possible, then recommend me the feasible driver that acts a bridge between Windows and SPI Flash memory to read file system data.

alexandrehagihara, 2023-04-17:
Hello @smitpatel30, good to know that I helped you. About the data transfer, it depends on your hardware design. I think that there's no specific driver to do what you need.
If you are able to dump your SPI Flash Memory to a raw binary file in your PC, you can use the python wrapper [littlefs-python](https://pypi.org/project/littlefs-python/) to mount this memory dump and then manipulate the file system programatically, reading the file content from the Python LittleFS class and write the content into another file in your PC.
If you inspect the 'examples' folder of its github page https://github.com/jrast/littlefs-python, theres a file 'walk.py' that can be used as a starting point, because it shows how you can do this mount process (basically open the raw memory dump file into the LittleFS context buffer and then call the .mount() function, but you should also take care of using the correct settings to mount it in the same way you do with your microcontroller)
Another way is to run a code that open your file with the microcontroller and write the content to a serial port connected to your PC and then you copy-paste the data into a file in your PC.

noomio, 2023-04-18:
Hi,

If you want to transfer files to PC you need to use a file transfer protocol.
There are a few but I have integrated kermit and push/pull files that way.

You need to port/add ekermit on your platform and expose it to serial or USB and invoke commands from you PC.

It works really well. I transfer MB of data this way from SPI flash.

smitpatel30, 2023-04-19:
> Hi,
>
> If you want to transfer files to PC you need to use a file transfer protocol. There are a few but I have integrated kermit and push/pull files that way.
>
> You need to port/add ekermit on your [...]

> Hi,
>
> If you want to transfer files to PC you need to use a file transfer protocol. There are a few but I have integrated kermit and push/pull files that way.
>
> You need to port/add ekermit on your [...]

smitpatel30, 2023-04-19:
Hi, @noomio for your advise.. Can you give the roadmap of transfering files from SPI Flash memory to PC via a Microcontroller.
I am a newbie in this field. I hadn't used FTP(File Transfer Protocol) in my past. How can I use this protocol to send the data from Flash Memory to PC ????....

closed 2023-05-02 by smitpatel30