littlefs-project/littlefs #797: Little FS Reading the same addresses over and over, causing 1min+ delay in program

CalebDueck, 2023-04-12:
Is it expected behavior that littlefs reads the same memory locations over and over?

Shortly after startup of my program it reads a tremendous amount of memory addresses. I assume this is to load the memory into RAM so that it doesn't need to read the memory addresses again. However, it will sometimes read the same pages multiple times in a row. See "lfs infinite read loop.txt"
It continues reading like this for a few minutes (normally our state machine takes 15 seconds to run so a delay like this just for our memory system is unacceptable).

Why does this occur and how could I resolve this?
It doesn't happen exclusively at the start of the program.

Any help would be appreciated.

[lfs infinite read loop.txt](https://github.com/littlefs-project/littlefs/files/11215005/lfs.infinite.read.loop.txt)
[continues after infinite read.txt](https://github.com/littlefs-project/littlefs/files/11215006/continues.after.infinite.read.txt)

gte1, 2023-04-14:
I have seen such huge delays too (32MB NOR flash). Partition has 16+ directories, each with 50 files. The delay typically happens when one overwrites a file already there (e.g. when uploading files via Zephyr mcumgr). Maybe there is some garbage collection going on?

geky, 2023-04-14:
This looks like it may be reflecting the same underlying issue in https://github.com/littlefs-project/littlefs/issues/783.

It's something I'm looking into rather deeply at the moment.  But also a rather deeply rooted issue in littlefs's design.

CalebDueck, 2023-04-14:
Thank you for sharing that other thread.
I may try limiting the metadata_max as currently our need for speed is greater than our need for memory.
You mentioned in your initial reply 
"littlefs generally always benefits from a smaller program size."
How would you define "smaller"?
Would this be controlled through my calls to lfs_write, or is there a setting I should tweak to make it program in smaller chunks?

geky, 2023-04-15:
Let us know if metadata_max works for your case. I think it may be the best way to work around performance issues in littlefs at the moment.

> You mentioned in your initial reply
> "littlefs generally always benefits from a smaller program size."
> How would you define "smaller"?

Ah, I can see how this could be confusing.

The "program size" in that sentence was referring to the `prog_size` config option, which controls how littlefs talks to the low-level block device functions. If you're using something like Zephyr this may already be taken care of for you.

The `prog_size` determines the minimum size of program operations the underlying block device supports. littlefs will still program more data than `prog_size`, it will just make sure to add padding so the data is aligned to `prog_size` boundaries. Most NOR flash devices set both `read_size`/`prog_size` to 1 byte.

At the user level, the size you send to `lfs_file_write` generally doesn't matter. It's generally better to send large writes to reduce the total number of operations, but littlefs will collect writes into each file's cache until the cache is full or `lfs_file_sync/close` is called, and only then write the data to disk.

CalebDueck, 2023-05-25:
Setting metadata_max to 1 page rather than the default of 1 block (64 pages) significantly sped up writing to littlefs.
Thanks.

geky, 2023-12-19:
Hi @qiumikee, that is an interesting observation.

I should add a bit to my comment. The "generally" in "littlefs generally always benefits from a smaller program size" is doing a lot of work. In a perfect world this would be true for both read_size/prog_size, but algorithm limitations make that not always possible.

I looked through the mdir functions and I think the most likely culprit is `lfs_dir_get`/`lfs_dir_getslice`. The core problem is looking up metadata in our mdirs requires reading backwards, and our caching layer only caches forwards.

The caching layer is doing something probably, I suspect you'll see no difference between 1 byte and 4 byte read sizes, but we can't provide a larger hint than that in `lfs_dir_get`/`lfs_dir_getslice` because of how it reads backwards.

This could probably be fixed, but there's already significant work going on to replace the mdir data structure with a tree, so I don't think it will be a priority...

And, fun thing about the tree rewrite, I've looked at applying caching to a tree, but I think the problem is equivalent to writing a branch predictor. Which is probably out of scope for littlefs.

What increasing `read_size` is doing here is forcing littlefs to read in larger chunks. This happens to be good for the backwards reads when metadata is "small".

---

I think the main takeaway is anything I say here is only a rule of thumb/good starting point. You should benchmark on your own device/configuration/use-case if performance is important.

If your files are larger than 4B, the backwards metadata reads will be sparse, so larger read_size may penalize you by forcing more data to be read to satisfy the read-alignment, for example (or maybe not, it's hard to know exactly).