littlefs-project/littlefs #803: Corrupted dir pair at {0x0, 0x1} LFS_ERR_NOSPC and  LFS_ERR_CORRUPT

Aravind0812, 2023-04-18:
1. I am currently working on implementing LittleFS on the W25Q128JW through QSPI.  
I am writing data to this file less than cache size  I am able to close the file successfully.my data size equal to or above cache size it return -84 (0x1 block unwritable ) (corrupted).some time i get  LFS_ERR_NOSPC       = -28,  // No space left on device
2. Then  check  i write first 256 bytes to a file I am able to close the file successfully .Again i open the file i write 256 bytes to a file i get error like
3.i check sync() , flush() but i get error how to solve that problem
Operation result: -84
../Core/Src/lfs.c:3088:debug: Bad block at 0x34a
                                                BUSY...
No longer BUSY...
Operation result: -84
../Core/Src/lfs.c:3088:debug: Bad block at 0x34b
                                                BUSY...
No longer BUSY...
Operation result: -84
../Core/Src/lfs.c:3088:debug: Bad block at 0x34c......

This is my configuration:
 /* Block device configuration */

	read_size   = 64
       prog_size   = 64
       block_size  = 4096
       block_count = 4096
       cache_size  = 512
       lookahead_size = 64
       block_cycles = 500

evpopov, 2023-04-21:
I'd suggest you start by modifying your .sync() .prog() and .read() functions to do the following:
When .sync() is called, do nothing and return LFS_ERR_OK.
When .read() is called, print out the block number that it is given, and the actual address you read from. Also, for 1 call of .read() perform the serial flash read 2-3 times and compare the results before returning LFS_ERR_OK. This ensures that your hardware/driver is providing consistent data to LittleFS
When .write() is called, print out the block number that it is given, and the actual address you write to. Perform the write, then read back the region at least once and verify the write was successful before returning LFS_ERR_OK. This ensures that whatever LittleFS asks you to write, you actually write.

That should get you going in the right direction because what you are describing as far as writing goes is very simple and works just fine. I wouldn't suspect LittleFS having a bug as simple as that at all, so triple-check your porting layer.

evpopov, 2023-04-21:
Sorry, forgot to mention one more check you should perform....
Your block size is 4k, but your prog_size is 64 bytes. That means that you should be able to do the following:
```
lfs_block_t block 1234;
lfs_config.prog( &lfs_config, block , 0, &data_1, 64);
lfs_config.prog( &lfs_config, block , 64, &data_2, 64);
lfs_config.prog( &lfs_config, block , 128, &data_3, 64);
```
and expect that data_1 is still at offset 0 of block 1234.
What I'm trying to say is: Make absolutely sure that a call to .prog() does not erase the block and rather just programs the data at the offset given.

closed 2023-04-24 by Aravind0812