littlefs-project/littlefs #813: Little FS on the GPX2 Cubesat

keck-in-space, 2023-04-27:
Hi @geky,

I just wanted to let you know that LFS is being used on the GPX2 cube satellite and I thought you might find that interesting.

https://drive.google.com/file/d/1ER7yYcAwVRhB8uA72tkLbAGZbN0A8Pr_/view?usp=share_link Here's a link to the presentation. LFS is mentioned on slides 7 and 8. The video of the presentation should be uploaded later next month.

Thanks for all your hard work on LFS! I'll go ahead and close this issue.

closed 2023-04-27 by keck-in-space

geky, 2023-04-30:
Hi @keck-in-space, this is very cool! Thanks for sharing. And congrats on what looks like a successful mission.

Those are some beautiful images, looks like I have a new desktop wallpaper.

Is there interest in sharing this with a wider audience, i.e. adding GPX2 to a list of projects that use LittleFS?

keck-in-space, 2023-05-01:
Thank you, @geky! Glad to hear you enjoyed the images! If you want, I'll update this issue once the presentation is edited and put on YouTube.

The main wrinkle with sharing this more widely is to ensure the footnote on the slides is included (reproduced below). The information in the slide deck is released to the public, however, so feel free to share it if desired. 

> Neither the U.S. Government nor NASA endorse or recommend any commercial products, processes, or services. Reference to or appearance of any specific commercial products, processes, or services by trade name, trademark, manufacturer, or otherwise, in NASA materials does not constitute or imply its endorsement, recommendation, or favoring by the U.S. Government or NASA.

geky, 2023-05-04:
Makes sense, I'll let you know if/when I add a link anywhere and make sure to include this text.

Hmm, do you know if NASA discourages references to missions in external/unaffiliated project's promotional material? Adding a link here would probably fall into that category and I wouldn't want to put you in an awkward position.

keck-in-space, 2023-05-04:
Let me ask around and get back to you! I've been taking the most conservative route, but the information is public and was approved for release so you should be able to link to it and discuss it. Any further internal information would need to be approved, but the slideshow and associated comments during the presentation are approved for public release and use. I think that's the purpose of sharing with the community -- show what we learned and what we did, and how it worked out.

keck-in-space, 2023-05-05:
I checked internally, and you are allowed to use the information in the presentation and reference that GPX2 utilized LFS. You can also link to the presentation since this is considered public information. Thanks again for all your hard work on LFS! As you'll hear in the presentation, it has worked very well and continues to work well on orbit.