littlefs-project/littlefs #815: Bad and Superblock error at the closing of file

smitpatel30, 2023-05-01:
Dear, Developers.

Hope are you doing good. I had created a file system in the W25Q16JV SPI Flash memory for saving the timelog data. Now, the time when I created proper , it was working marvellously  After some days, when I came back to this project, I found an error while closing the file..

Everytime I write the data , it shows error like bad sand superblock at x1.

What is the ultimate solution for get rid out of this problem??........

Thanks for reading this...

geky, 2023-05-01:
Hi @smitpatel30 thanks for creating an issue, unfortunately there's not really enough information to know what the problem could be.

One common reason for the bad superblock error is a mistake mapping littlefs to the block device. This might be the first time littlefs tried to use block 1, so if the block device layer doesn't write block 1 correctly littlefs will report it as a bad block.

It may be worth writing some tests that your block device works. I have some tests here that may be useful as a starting point: https://github.com/littlefs-project/littlefs/blob/devel/tests/test_bd.toml

smitpatel30, 2023-05-02:
Due to bad block error, my SPI Flash, named W25Q16JV, was completely blocked or unwritable .. I have replaced this flash memory with the another one. I am able to write write and read data in this flash without mounting LittleFS. Now, when I mounted the file system and created a file .. I am able to store data in ".txt" or ".csv" file only first two times. When I enter the data third time, it shows me the error of bad block and error code of -28..(that means no space).  But , how can two strings occupies the whole memory. I have been facing this error  since a week. Earlier, It was running spectacularly...

I am always facing an error at the time of closing of file.. Is their any solution for file closing error??....

smitpatel30, 2023-05-02:
Here are the configurations for W25q16JV , SPI Flash for mounting with LittleFS.. Let me know if there is something wrong??..

With this, Kindly provide your contact details such as insta Id, phone number, whatsapp number, or anything else that allows easier communication with us...
Review the below screenshots.. please tell if I doing some with block parameter such as  lookahead size, cache size or anything else....

[image]
[image]