littlefs-project/littlefs #820: lfs_util.h includes non-libc <sys/types.h> that is unavailable with Arm Compiler for Embedded

LDong-Arm, 2023-05-05:
In https://github.com/littlefs-project/littlefs/blob/ec3ec86bccd619903c57dc650af31e74dd5a4888/lfs_util.h#L26

`<sys/types.h>` is not part of the standard libc. When using Arm Compiler for Embedded, this header is not found.

This `#include` was added in https://github.com/littlefs-project/littlefs/commit/61455b6191bb5da9d8b799f4b7056570d3df8820 for testing. But removing this include does not prevent `make test` from working. I built and ran the test on Ubuntu 22.04 with gcc 11.3.0. Maybe this include is not needed?

@geky

vvn-git, 2023-05-16:
I also removed it from v2.6 (ARMCC 5.0.x).

geky, 2023-05-17:
Hi @LDong-Arm, @vvn-git, thanks for creating an issue. I went ahead and created a PR to remove this: https://github.com/littlefs-project/littlefs/pull/832, will bring this in on the next patch release.

sys/types.h was probably added at some point for access to `ssize_t` (a strangely missing type in C's stdlib), but it's not currently needed and I can't really think of a case where it would be needed in the future.

closed 2023-05-24 by LDong-Arm