littlefs-project/littlefs #822: Does littlefs require the order in which program calls are executed between sync calls to be consistent?

MFaehling, 2023-05-09:
Context: I'm currently trying to use LittleFS on an eMMC flash. With this flash, it is possible to use a cache that stores data in between flushes(I would flush the eMMC with each `sync` call). However, this way the order of the writes between flushes is not guaranteed.

So does the power loss resiliency of LittleFS still work when the order in which `program` calls between `sync` calls are committed to flash is not guaranteed?

geky, 2023-05-31:
Hi @MFaehling, thanks for raising an issue.

Good question. Unfortunately with the way LittleFS currently uses sync, I think unordered programs break power-loss guarantees. But this may easy to fix.

Currently programs maintain order, sync is just used to allow users to know when any in-flight progs have completed. (The original consideration was devices with maybe one block of caching).

---

At a high-level, LittleFS is a set of metadata logs containing pointers to copy-on-write (CoW) data structures.

Currently, we only call sync at the point we finish writing the metadata+checksum to the log. The idea is this provides guarantees to the caller of, say, lfs_file_sync, that the data is written in case ordering is needed between LittleFS and other persistent hardware.

But having a single sync call is a problem. We don't look at the CoW data structures when determining if a metadata update has completed, so if CoW writes are reordered to after the metadata writes, and power is lost, the filesystem will be broken.

Fortunately I think the fix may be quite easy. If we have pending CoW data structures, we need to call sync twice:
1. Before starting the metadata write.
2. After the metadata write is completed.

Thanks to the checksum, the order of progs within the metadata log don't matter. The checksum is only valid when all dependency progs have completed. 

I'm going to take a look at what some of the other filesystems do here. There may be some other subtleties we're missing.

geky, 2024-02-27:
Hi @MFaehling, sorry about leaving this unresolved for so long.

On second inspection this really should have been treated as a bug in littlefs.

https://github.com/littlefs-project/littlefs/pull/948 should be fixing this by calling bd `sync` in the correct places, but let me know if anything looks wrong.