littlefs-project/littlefs #823: lfs_cache_zero cause corruption

admin456dnd, 2023-05-10:
The lfs_cache_zero will cause corruption if cachesize !=program_size .

static inline void lfs_cache_zero(lfs_t *lfs, lfs_cache_t *pcache) {
    // zero to avoid information leak
    memset(pcache->buffer, 0xff, lfs->cfg->cache_size);
    pcache->block = LFS_BLOCK_NULL;
}

The lfs->cfg->cache_size is error here when prog_size != cache_size the memset function will set part of memory that not belong to it to 0xff, this will cause secrious problem.
I have tested the bug on my board( stm32 + littlefs 2.6)
so is that expected behavior cache size should be = or <= prog_size?

MFaehling, 2023-05-11:
See the comment above cache_size https://github.com/littlefs-project/littlefs/blob/ec3ec86bccd619903c57dc650af31e74dd5a4888/lfs.h#L222-L223C16

> `Must be a multiple of the read and program sizes, and a factor of the block size.`

admin456dnd, 2023-05-12:
Yes i read already. In my case program and read size is 512 bytes when i set cache size as 1024(i think this multiple of prog and read size) it will cause corruption.

geky, 2023-05-17:
I think the confusion here is that the prog_buffer/read_buffer need to be set to the cache_size. Aka `sizeof(prog_buffer) == cache_size`.

The prog_size indicates the minimum size of what can be physically written during a prog operation, and is often 1-byte on NOR flash.

I realize the naming of these options is confusing, and it's something I want to change moving forward, but since it's a part of the API changing these names would break existing code, and it's not worth a major version bump on its own.