littlefs-project/littlefs #825: Disk format migration considerations

rojer, 2023-05-14:
> This is the first on-disk minor version bump for littlefs, and is untested waters, so do open a GitHub issue if this creates any problems.

let me describe a problem this creates for us. our devices have 2 slots, each with an app and a corresponding filesystem.
our firmware image carries its own "base" filesystem image and during first boot mounts the other slot's FS (read-only) to copy certain volatile configuration files onto its own base image, thus carrying over user's configuration and customizations.

this works perfectly for upgrades - we can ship major filesystem upgrades, even entirely changing filesystem, so long as the newer fw supports both (we have migrated form spiffs to lfs this way).
however, this also requires that older firmware also supports both, in case a downgrade is required (which sometimes happens).
automatic bumping of disk format to 2.1 in 2.6 thus presents a problem for us: during a downgrade, older firmware will be unable to mount newer firmware's fs to transfer config files, which would mean downgrade is not possible.

now, i also understand that disk format changes happen and thing need to move on. but we need to create a window of releases that can handle both 2.0 and 2.1 formats, and automatic bump makes it impossible.
what we need is a "2.1-capable but still using 2.0" mode for new releases, and our plan would then be be to ship a number of releases in that mode, before finally moving to 2.1.

alternatively, a concept of "disk format for reading" vs "disk format for writing" could be introduced. my understanding is that 2.0 could actually be read by lfs 2.5, which would suit us just fine, as we mount the fs read-only.

geky, 2023-05-17:
Hi @rojer, thanks for creating an issue. These compatibility-upgrade problems are hard to predict.

This is an interesting dilemma.

If I understand the problem correctly, what you need is a version of littlefs that can read both lfs2.0 and lfs2.1, but only writes lfs2.0. This sounds doable. lfs2.1 can read lfs2.0, so for lfs2.0->lfs2.1 this would just mean not writing out the fcrc tags. The main annoyance would be adding the minor_version as a config option so littlefs knows what to do.

I'm also unsure if this will be as easy for future on-disk minor version bumps, it's hard to predict when in a perfect world we will never need to bump the minor version again.

I suppose with https://github.com/littlefs-project/littlefs/issues/826 (user-readable minor version), it would be possible to implement the migration as you did for spiffs<->lfs, though at the cost of having two filesystems.

geky, 2023-06-07:
Working on this here: https://github.com/littlefs-project/littlefs/pull/839

I was thinking about what else could be done, but I think the best option is to just add a config option to littlefs to indicate which version to write with.

With https://github.com/littlefs-project/littlefs/pull/839, (unless it changes), littlefs should be able to read both lfs2.1/lfs2.0 and write either lfs2.1 or lfs2.0 depending on `cfg.disk_version`.

rojer, 2023-07-10:
Hi @geky , sorry for not replying earlier - notifications got lost in the noise.
Yes, this is exactly what we need and it works great, thanks a lot!

closed 2023-07-10 by rojer