littlefs-project/littlefs #828: Reading metadata block in "commit order", and "reverse commit order"?

test-sg, 2023-05-15:
I am a beginner to LittleFS, and I want to learn it.
I refer to:
https://github.com/littlefs-project/littlefs/issues/333

Basically I know that the metadata tag is XORED, just like XORED linked-list. I understand how it works, ie. user is able to read from either direction:

[image]

However, I do not understand the use case of reading the metadata blocks in either direction. What's the purpose? Are they any example?

When reading in commit order, what does it mean by "We need to read "in-order" to be able to fall back to a previous commit if we find a bad CRC."?

geky, 2023-05-17:
Hi @test-sg,

Consider an example where we rewrite a file several times. Since our metadata blocks are append-only, at a high-level our on-disk metadata may look something like this:

```
.-------------------. |
| file A.txt="aaaa" | |
|-------------------| |
| file A.txt="bbbb" | | new commits are appended
|-------------------| | in this order
| file A.txt="cccc" | |
|-------------------| |
| file A.txt="dddd" | |
|-------------------| v
|   erased bytes    |
|                   |
|                   |
|                   |
|                   |
|                   |
'-------------------'
```

When we first access a metadata block, for example during mount, we have no information about the metadata block so we don't know where exactly the end is. To find the end, we need to start at the beginning, scanning tags, and checking for valid checksums. When we hit an invalid checksum we assume that is the current end of the metadata block.

But we don't really care about the tags at the beginning of the metadata block. Since the file has been overwritten, most of the earlier tags such as `A.txt="aaaa"` are out-of-date. We only care about the most recent tag `A.txt="dddd"`. Scanning backwards allows us to stop after the first `A.txt` we find. This is a slight performance improvement, but it's interesting to note this doesn't improve the runtime vs scanning from the front. Both run in $O(n)$.

---

It gets a bit into the weeds, but in littlefs we need to be able to search through tags backwards for a functional reason.

It's difficult to work with file names in a RAM-bounded system, so littlefs maps each file to a metadata-block-specific id. These ids represent the file's position in an imaginary array stored in the metadata block.

Consider some files:

```
metadata log                  imaginary array
.-------------------.  
| create id0=A.txt  |  .     
|-------------------|  |
| data id0="aaaa"   |  *      [A.txt="aaaa"]
|-------------------|  |
| create id1=B.txt  |  | .         
|-------------------|  | |
| data id1="bbbb"   |  | *    [A.txt="aaaa", B.txt="bbbb"]
|-------------------|  | |
| create id2=C.txt  |  | | .      
|-------------------|  | | |
| data id2="cccc"   |  | | *  [A.txt="aaaa", B.txt="bbbb", C.txt="cccc"]
|-------------------|  | | |
|         |         |  v v v
|         v         |  
|                   |  
|                   |  
'-------------------'
```

This is a simple case, but we keep the files in sorted order in our imaginary array. If we create the files out of order it might look something like this:

```
metadata log                  imaginary array
.-------------------.  
| create id0=A.txt  |  .     
|-------------------|  |
| data id0="aaaa"   |  *      [A.txt="aaaa"]
|-------------------|  |
| create id1=C.txt  |  | .
|-------------------|  | |
| data id1="cccc"   |  | *    [A.txt="aaaa", C.txt="cccc"]
|-------------------|  |  \
| create id1=B.txt  |  | . |
|-------------------|  | | |
| data id1="bbbb"   |  | * |  [A.txt="aaaa", B.txt="bbbb", C.txt="cccc"]
|-------------------|  | | |
|         |         |  v v v
|         v         |  
|                   |  
|                   |  
'-------------------'
```

And deletes make it more confusing:

```
metadata log                  imaginary array
.-------------------.  
| create id0=A.txt  |  .     
|-------------------|  |
| data id0="aaaa"   |  *      [A.txt="aaaa"]
|-------------------|  |
| create id1=B.txt  |  | .
|-------------------|  | |
| data id1="bbbb"   |  | *    [A.txt="aaaa", B.txt="bbbb"]
|-------------------|  | |
| delete id0        |  ' |    [B.txt="bbbb"]
|-------------------|   /
|         |         |  v
|         v         |  
|                   |  
|                   |  
'-------------------'
```

The way id creation/deletion influences neighboring ids creates some challenges. The main one being that the id of a file may change over time.

In order to perform lookups by id, we need to scan backwards, adjusting the id we are looking for as we come across create/delete tags.

As far as I'm aware it's not possible to do this while scanning forward, since we don't know which id we care about as we look at tags. But I encourage you to try, it's an interesting problem.

---

It's interesting to note this is only a problem when we want to keep these ids "dense". That is, we are able to map each id to an index of an imaginary array.

If our ids are "sparse", and we allow holes in the array, we don't need to shift neighboring ids as we create/delete. Thinking about it this way, file names are effectively a sort of "sparse" id, which makes it possible to scan for filenames forward.

And we actually do scan for filenames forward, since we can combine that scan with our metadata fetch to do both in one pass. Once we find a filename though, we internally keep track of it as a "dense" id and do subsequent lookups backwards.

test-sg, 2023-05-19:
> 

Dear geky,

Above is very good information for beginner like me; they are not found in LittleFS Github, are they? Besides LittleFS Github, where can I find good tutorial for LittleFS?

geky, 2023-05-23:
Sorry @test-sg, I'm not really aware of any other resource. There is documentation in https://github.com/littlefs-project/littlefs/blob/master/DESIGN.md and https://github.com/littlefs-project/littlefs/blob/master/SPEC.md, but where this isn't complete is on me.

Searching the issues may also lead to common questions.

littlefs is relatively new, with some glaring issues, and balancing filesystem improvements vs documentation is a bit difficult.