littlefs-project/littlefs #856: Can littleFS run with STM32 and external raw NAND Flash ?

ShaiBinary, 2023-07-12:
Can littleFS run with STM32 and external raw NAND Flash ?

kyrreaa, 2023-11-02:
If you write a NAND access layer for it to init the flash and provide the read, write and erase functions then LittleFS should not care what the underlying storage really is. This is what I have done for nRF5340 (SPIM access since QSPI was hardcoded for NOR flash).
You need to set up good values for the read-, write- and erase-blocks as well as cache though so the requests align with the flash and your implementation. (If you ONLY implement single sector writes for example, do not give LittleFS a larger cache or it will request multi-sector writes occasionally.)

geky, 2023-11-14:
I just wanted to comment that even if you limit LittleFS's cache, LittleFS may still write multiple-sectors in a single prog call. This can happen if you call `lfs_file_write` with a buffer larger than the cache, in which case LittleFS bypasses the cache.

Also LittleFS has a number of performance issues with NAND, particularly it does not scale well when block sizes become large. It may be worth looking into the issues labeled [performance](https://github.com/littlefs-project/littlefs/issues?q=is%3Aopen+is%3Aissue+label%3Aperformance).

kyrreaa, 2023-11-20:
Thank you @geky that is very useful to know. It also explains why I had some issues with my access layer before as I had done a naive implementation assuming only single block writes.