littlefs-project/littlefs #860: Write time increasing with time/file size

xgroleau, 2023-07-25:
Currently I have a long lived file and we are writing data for a long duration (multiple Mb). Though it seems the write operation becomes longer and longer as we add more data to the file and eventually it starves the MCU causing other operations to timeout. It starts at less than 1ms to write the file content to more than 100ms after adding a lot of data.

Note that we are writing a lot of small chunk of data (5 bytes) but we are using a cache for each file and it seems the number of operations (read/write accesses) increases with the length of the file.

Is this a known fact that the write operations are proportional to the file size? We are using NAND flash but adapted the driver to offer a similar interface to NOR flash so it shouldn't be an issue. Is there anyway to increase the speed? Like periodically sync the file to the filesystem?

LittleFS version: 2.5.1

xgroleau, 2023-07-25:
It seems buffering the writes quite dramatically. Not sure why since I am using a cache and I thought it should do it internally.

e107steved, 2023-07-26:
Are you doing random writes (rather than simply appending)? If so, that's your problem. The nature of the file structures mean that data after the insertion point has to be rewritten. Search the "issues"; there are some far better explanations of this than I can give.

xgroleau, 2023-07-26:
Hi, i'm doing sequential write. After some digging, I've found the info I was searching for in issue #783 so I'll close this issue. Buffering the write also helped quite a bit.

closed 2023-07-26 by xgroleau