littlefs-project/littlefs #869: How much memory is my lfs using

Dan2605, 2023-09-04:
Hi @geky, I want to calculate the current memory usage of the file system configuration I have set up. Is there a way to do this? I know of lfs_fs_size, but does this take into account of other "hidden" memory usage, such as maybe from the superblocks, unused blocks etc?

spresg, 2023-09-14:
Hi @geky 
How to find out the actual memory used?
Can I use lfs_stat()?

geky, 2023-09-14:
This should get you the usage in bytes, assuming littlefs is mounted and assuming you know the block_size of your device:

``` c
lfs_ssize_t blocks = lfs_fs_size(&lfs);
if (blocks < 0) {
    return blocks;
}

printf("usage: %d B\n", blocks * BLOCK_SIZE);
```

I considered adding this information to `lfs_fs_stat`, but `lfs_fs_size` has the potential to take a long time on larger filesystems, so I am hesitant. We can always add the field to `lfs_fs_stat` if it's useful in the future.