littlefs-project/littlefs #871: Failing to create a directory

sseoud, 2023-09-08:
Hi,
We are using littleFs with the following configuration
truct lfs_config cfg = {
  // block device configuration
  .read_size = 16,
  .prog_size = 256,
  .block_size = 4096,
  .block_count = 8,
  .cache_size = 256,
  .lookahead_size = 16,
  .block_cycles = 500
};
We are using the internal flash of the mcu.
I am trying to create 4 directories.
the first 3 are created successfully while the last on failing with error code -28
Now if I change the block_count to 16 then all the directories are created and everything is fine.
Can anyone please help me understand we I need to much space to create these directories
Thanks in advance

geky, 2023-09-10:
Hi @sseoud,

In the current design of littlefs, each directory gets its own metadata log. Each metadata log takes at minimum two blocks, so each directory needs 2 blocks. This includes the root.

(1 root + 4 directories)*2 = 10 blocks, so that would be why you see LFS_ERR_NOSPC ([-28](https://github.com/littlefs-project/littlefs/blob/130790fa915d104b5ca19524e86d9618fdcac848/lfs.h#L83)) when trying to create the last directory.

Using many directories with internal flash may not be a great fit in littlefs. It may make more sense to prefix files with application-specific prefixes as a part of the filename if you're only dealing with a handful of blocks.

---

It's not helpful, but I thought worth noting I have some plans in the works to remove this strict dir <-> log relationship. But it will take some time before it's usable since it's a part of some larger changes.

sseoud, 2023-09-11:
Hi @geky,
Thanks for the answer.
Is there a minimum requirement for the block size ?
If I set the block size to 256 (the smallest erasable size in my internal flash), I can't format the device.
Regards

geky, 2023-10-18:
Hi @sseoud, sorry for the late response,

There is a fundamental limit at 128 bytes ([assert here](https://github.com/littlefs-project/littlefs/blob/f77214d1f0a8ccd7ddc7e1204fedd25ee5a41534/lfs.c#L4141-L4142)). But littlefs also expects file/superblock metadata to fit in a single block. When you go below ~512 bytes this gets a bit iffy and isn't well tested.

You can try using smaller filenames or custom attributes, though I would expect NOSPC/RANGE/NAMETOOLONG errors if that is the issue.

I would also double check the erase size, 256 bytes is a bit surprising for flash, and the chip may have other requirements to use the smaller erase size. Out of curiosity are you able to share which chip you are using?