littlefs-project/littlefs #876: [QUESTION] Will seeking to position 0 clear the file?

TjoBitDk, 2023-09-19:
Hi

If I open a file and seek to start of the file and start writing. How will that affect the data allready in the file?

Example: I have a 200 byte file. I open it and seek to position 0 and writes 100 bytes.
Is the last 100 bytes stil in the file or will they be carbage collected at some point?

Thomas

geky, 2023-09-21:
Hi @TjoBitDk, littlefs tries to follow POSIX/BSD/Linux ([example](https://pubs.opengroup.org/onlinepubs/007904975/functions/open.html)) where possible, so when you open a file for writing the previous content will still be there. The contents won't mysteriously disappear or be garbage collected underneath you.

If you want to clear the previous data you can add `LFS_O_TRUNC` to your file open flags, or call `lfs_file_truncate` explicitly.

Also, unless you pass `LFS_O_APPEND` in your open flags, a file opened for writing starts at position 0 so you may not need an extra seek call.

hattesen, 2023-11-02:
@geky I think it would be super helpful if all issues labeled `question` would be added to a `FAQ.md` document, which could be referenced from `README.md`. It would also allow those `question` issues to be closed.

Let me know if you'd like me to assist and create a Pull request

geky, 2023-11-14:
Hi @hattesen, curating a FAQ would be greatly appreciated, though I don't think this repo is quite the right place for it.

I've been wanting to make a wiki of sorts in the littlefs org, with each page a simple markdown file, but haven't been able to get to it. GitHub wikis are a bit cumbersome and as far as a I can tell best a fork of a separate repo. But in the future said repo could also be decorated into a website.

If  you want to make an FAQ.md, I can figure out a home for it. Or if you want to wait for a home I can update this issue when a wiki/website is created.