littlefs-project/littlefs #883: littlefs writes only to the first erasable sector until this sector becomes full, corrupting its own filesystem. Unable to format the drive if prog_size=read_size=cache_size<64 Bytes.

pawelstraszak, 2023-10-20:
## Hardware
[STM32L496ZGT](https://www.st.com/en/microcontrollers-microprocessors/stm32l496zg.html)<==QSPI==>[W25Q80DV](https://www.winbond.com/hq/product/code-storage-flash-memory/serial-nor-flash/?__locale=en&partNo=W25Q80DV). I use [this development board](https://kameleonboard.org/).

## Boot count demo application
I am running a "boot counter demo" application, in which during every reboot, boot count value is read from a file, placed in a local variable on a stack, incremented, written-back to the same file (the file is overwritten) and, finally, a message containing the boot count value is sent via a serial port. Boot count value is of an uint32_t type (4 bytes long). Reboot is generated by resetting the MCU or interrupting the power supply (I tested both methods and both seem to work well - correct boot count value is held in the flash memory as expected).

## What is wrong

1) I am unable to format the drive when prog_size=read_size=cache_size<64 Bytes[^1].
2) Formatting succeeds only if I set the prog_size=read_size=cache_size>=64 Bytes.[^2]
3) Application seems to work as expected. Every time I reset the MCU or turn-off-and-on a power switch, the boot count is incremented (hooray!). But...
4) When overwriting the file containing the boot count, littlefs uses only one memory erasable-sector (is this an expected behavior?)
5) When this sector is full, littlefs searches for a free space only in this sector over-and-over (littlefs goes crazy).
6) Then littlefs performs a "self-destruction" and corrupts the filesystem.
7) After the "self-destruction", drive needs to be re-formatted.
8) After re-formatting the drive, everything happens one more time (goto: point 3).

[^1]:Very rarely and randomly, it is possible to format the drive with prog_size=read_size=cache_size=32 Bytes. But just after the only sector used by littlefs becomes full, littlefs corrupts the filesystem and even re-formatting of the drive fails.
[^2]:When prog_size=read_size=cache_size=256 Bytes, filesystem can be mounted, but sometimes file cannot be opened.

## Memory characteristics and littlefs configuration

### W25Q80DV characteristics:

- Capacity: 1 MiB = 1,048,576 Bytes
- Sector-size (min-erase-size): 4 kiB = 4,096 Bytes
- Page-size (max-program-size): 256 Bytes
- program 1 Byte .. 256 Bytes of data at a time

### With the 1st configuration drive formatting usually fails (32 bytes):

```C
const struct lfs_config cfg = {
      .read  = fs_flash_read,
      .prog  = fs_flash_prog,
      .erase = fs_flash_erase,
      .sync  = fs_flash_sync,

      .read_size = 	32,    // cache_size = read_size = prog_size [Bytes]
      .prog_size = 	32,    // cache_size = read_size = prog_size [Bytes]
      .block_size = 	4096,  // = erasable_sector_size [Bytes]
      .block_count = 	256,   // = flash_capacity / erasable_sector_size
      .cache_size = 	32,    // cache_size = read_size = prog_size [Bytes]
      .lookahead_size = 64,    // = block_count/8 (minimum, must be a multiple of 8) [Bytes]
      .block_cycles = 	100    // [Cycles]
};
```
Very rarely and randomly, however, it is possible to create a valid filesystem with the 1st configuration. But always exactly after every 124 or 125 (I don't remember the exact value) consecutive successful write operations, filesystem corrupts itself and even re-formatting fails.

```
Mounting the drive
../Drivers/littlefs/lfs.c:1346:error: Corrupted dir pair at {0x1, 0x0}
Error: -84
Attempting to re-format the drive:
../Drivers/littlefs/lfs.c:2049:debug: Bad block at 0x1
../Drivers/littlefs/lfs.c:2054:warn: Superblock 0x1 has become unwritable
Error: -28
Unable to re-format the drive
```

### With the 2nd configuration drive formatting usually succeeds (64 bytes):

```C
const struct lfs_config cfg = {
      .read  = fs_flash_read,
      .prog  = fs_flash_prog,
      .erase = fs_flash_erase,
      .sync  = fs_flash_sync,

      .read_size = 	64,    // cache_size = read_size = prog_size [Bytes]
[... 497 lines cut ...]
...
```

## What I have tested so far

The problem may lay in the:
- hardware,
- timings,
- STM32 QSPI peripheral driver (STM32's HAL QSPI driver),
- W25Q80DV QSPI driver: qspi_xxxxxxx() utilizing STM32's HAL QSPI library,
- wrapper/glue functions for littlefs: fs_flash_read(), fs_flash_prog(), fs_flash_erase(), fs_init(),
- littlefs use in main.c,
- littlefs itself.

I tested addresses, block numbers and offsets during read/write/erase operations in glue/wrapper functions. It seems OK.

I also tested underlying low-level functions (qspi_xxxxxxx()). These functions are supplied by the vendor of the development board and internally call STM32 HAL QSPI library functions.

I made to them small improvements and bug fixes (because i.e. erase_sector function wasn't waiting for erase operation to complete, there was wrong ERASE_SECTOR opcode, etc.). 

I tested QSPI memory raw write, raw erase and raw readout. Before and after every test I uploaded whole affected memory to see what has changed. These tests prove that I can successfully write at an arbitrary address an arbitrary number of bytes (even if it requires programming multiple pages and slicing input data to fit to the page boundaries, because data are programmed in pages of 256 Bytes) without any issues. I tested also the erase function (whether it erases only the locations that are intended to be erased). It looks OK. By the way, if low-level functions were totally broken, littlefs would not have any chance to successfully format, mount, read and write data to the flash memory about 60 times...

## Conclusion

Probably, there is something bad happening with littlefs itself. It is very suspicious that only the first sector (4KiB) is used by littlefs. When it becomes full, erase operation is performed and the filesystem becomes corrupted. However, I do not definitively rule out errors beyond littlefs itself.

pawelstraszak, 2023-10-20:
Close this issue, cfg->prog_size and cfg->read_size should be replaced with cfg->block_size.

closed 2023-10-20 by pawelstraszak

mc-xgl, 2025-06-03:
> [@mc-xgl](https://github.com/mc-xgl), the problem was these two lines in and :`fs_flash_read``fs_flash_prog`
> 
>   address = block*cfg->read_size + off;
>   address = block*cfg->prog_size + off;
> They should be:
> 
>   address = block*cfg->block_size + off;
>   address = block*cfg->block_size + off;
> The reason is is in units of .`block``block_size`
> 
> When littlefs calls and , is guaranteed to be a multiple (>=) of /. But other than that, / only control internal caches as you mention.`fs_flash_read``fs_flash_prog``size``read_size``prog_size``read_size``prog_size`

@geky Thank you for your reply. It seems that the issue I encountered is not the same as yours. I also encountered the LFS_ERR_CORRUPT error in the file system when using Little (LFS_VERSION is 0x00020005). The phenomenon is as follows: after writing data to files in the third and fourth-level directories (with the highest reproduction probability, but it is not certain whether it is a reproduction condition), I performed a soft reboot. When opening the files again, I found that the file system was likely to be corrupted (with a reproduction frequency of about once a day). I am using the RT-THREAD dfs_lfs file directly as the FLASH driver, with SFUD as the underlying layer. I have tested the SFUD for full-chip read and write operations, and no issues were found. The data remains normal after reboot. The SFUD read and write interfaces have been tested for 3 days, so it is basically possible to eliminate hardware issues. Before reboot, I ensure that the APP calls the close interface. If the close operation can ensure synchronization, then the reboot will not interrupt file reading and writing. I have tried to lock the external FLASH before reboot to attempt to fix this issue, but unfortunately, the problem was not resolved. Therefore, it can be concluded that the problem is not caused by signal jitter during the reboot process.

In my configuration, the Little FS related settings are as follows:

```c
#define LFS_READ_SIZE 256
#define LFS_PROG_SIZE 256
#define LFS_BLOCK_SIZE 4096
#define LFS_CACHE_SIZE 256
#define LFS_BLOCK_CYCLES 800
#define LFS_THREADSAFE 1
#define LFS_LOOKAHEAD_MAX 128
```

Could you please help with the analysis? This issue has been bothering me for a month. Currently, based on the tests, an APP issue cannot be ruled out. However, I stubbornly believe that APP operations on Littlefs should not affect the file system itself. Therefore, this issue seems to be at an impasse at present.

Below are my FLASH read and write interfaces:

```c
// Read a region in a block. Negative error codes are propogated
// to the user.
static int _lfs_flash_read(const struct lfs_config* c, lfs_block_t block, lfs_off_t off, void* buffer, lfs_size_t size)
{
    struct rt_mtd_nor_device* mtd_nor;

    RT_ASSERT(c != RT_NULL);
    RT_ASSERT(c->context != RT_NULL);

    mtd_nor = (struct rt_mtd_nor_device*)c->context;
    if (rt_mtd_nor_read(mtd_nor, block * c->block_size + off, buffer, size) != size)
    {
        return LFS_ERR_IO;
    }

    return LFS_ERR_OK;
}

// Program a region in a block. The block must have previously
// been erased. Negative error codes are propogated to the user.
// May return LFS_ERR_CORRUPT if the block should be considered bad.
static int _lfs_flash_prog(const struct lfs_config* c, lfs_block_t block, lfs_off_t off, const void* buffer, lfs_size_t size)
{
    struct rt_mtd_nor_device* mtd_nor;

    RT_ASSERT(c != RT_NULL);
    RT_ASSERT(c->context != RT_NULL);

    mtd_nor = (struct rt_mtd_nor_device*)c->context;
    if (rt_mtd_nor_write(mtd_nor, block * c->block_size + off, buffer, size) != size)
    {
        return LFS_ERR_IO;
    }

    return LFS_ERR_OK;
}

// Erase a block. A block must be erased before being programmed.
// The state of an erased block is undefined. Negative error codes
// are propogated to the user.
// May return LFS_ERR_CORRUPT if the block should be considered bad.
static int _lfs_flash_erase(const struct lfs_config* c, lfs_block_t block)
{
    struct rt_mtd_nor_device* mtd_nor;

    RT_ASSERT(c != RT_NULL);
    RT_ASSERT(c->context != RT_NULL);

    mtd_nor = (struct rt_mtd_nor_device*)c->context;
    if (rt_mtd_nor_erase_block(mtd_nor, block * c->block_size, c->block_size) != RT_EOK)
    {
        return LFS_ERR_IO;
    }

    return LFS_ERR_OK;
}
```

In the interface, the concept of a block is a sector, which is the minimum erasable unit of the W25Q128 chip, with a size of 4096. The write buffer of the W25Q128 is only 256 bytes.

mc-xgl, 2025-06-03:
> > [@mc-xgl](https://github.com/mc-xgl), the problem was these two lines in and :`` fs_flash_read``fs_flash_prog ``
> > address = block_cfg->read_size + off;
> > address = block_cfg->prog_size + off;
> > They should be:
> > address = block_cfg->block_size + off;
> > address = block_cfg->block_size + off;
> > The reason is is in units of .`` block``block_size ``
> > When littlefs calls and , is guaranteed to be a multiple (>=) of /. But other than that, / only control internal caches as you mention.`` fs_flash_read``fs_flash_prog``size``read_size``prog_size``read_size``prog_size ``
> 
> [@geky](https://github.com/geky) Thank you for your reply. It seems that the issue I encountered is not the same as yours. I also encountered the LFS_ERR_CORRUPT error in the file system when using Little (LFS_VERSION is 0x00020005). The phenomenon is as follows: after writing data to files in the third and fourth-level directories (with the highest reproduction probability, but it is not certain whether it is a reproduction condition), I performed a soft reboot. When opening the files again, I found that the file system was likely to be corrupted (with a reproduction frequency of about once a day). I am using the RT-THREAD dfs_lfs file directly as the FLASH driver, with SFUD as the underlying layer. I have tested the SFUD for full-chip read and write operations, and no issues were found. The data remains normal after reboot. The SFUD read and write interfaces have been tested for 3 days, so it is basically possible to eliminate hardware issues. Before reboot, I ensure that the APP calls the close interface. If the close operation can ensure synchronization, then the reboot will not interrupt file reading and writing. I have tried to lock the external FLASH before reboot to attempt to fix this issue, but unfortunately, the problem was not resolved. Therefore, it can be concluded that the problem is not caused by signal jitter during the reboot process.
> 
> In my configuration, the Little FS related settings are as follows:
> 
> #define LFS_READ_SIZE 256
> #define LFS_PROG_SIZE 256
> #define LFS_BLOCK_SIZE 4096
> #define LFS_CACHE_SIZE 256
> #define LFS_BLOCK_CYCLES 800
> #define LFS_THREADSAFE 1
> #define LFS_LOOKAHEAD_MAX 128
> Could you please help with the analysis? This issue has been bothering me for a month. Currently, based on the tests, an APP issue cannot be ruled out. However, I stubbornly believe that APP operations on Littlefs should not affect the file system itself. Therefore, this issue seems to be at an impasse at present.
> 
> Below are my FLASH read and write interfaces:
> 
> // Read a region in a block. Negative error codes are propogated
> // to the user.
> static int _lfs_flash_read(const struct lfs_config* c, lfs_block_t block, lfs_off_t off, void* buffer, lfs_size_t size)
> {
>     struct rt_mtd_nor_device* mtd_nor;
> 
>     RT_ASSERT(c != RT_NULL);
>     RT_ASSERT(c->context != RT_NULL);
> 
>     mtd_nor = (struct rt_mtd_nor_device*)c->context;
>     if (rt_mtd_nor_read(mtd_nor, block * c->block_size + off, buffer, size) != size)
>     {
>         return LFS_ERR_IO;
>     }
> 
>     return LFS_ERR_OK;
> }
> 
> // Program a region in a block. The block must have previously
> // been erased. Negative error codes are propogated to the user.
> // May return LFS_ERR_CORRUPT if the block should be considered bad.
> static int _lfs_flash_prog(const struct lfs_config* c, lfs_block_t block, lfs_off_t off, const void* buffer, lfs_size_t size)
> {
>     struct rt_mtd_nor_device* mtd_nor;
> 
>     RT_ASSERT(c != RT_NULL);
>     RT_ASSERT(c->context != RT_NULL);
> 
>     mtd_nor = (struct rt_mtd_nor_device*)c->context;
>     if (rt_mtd_nor_write(mtd_nor, block * c->block_size + off, buffer, size) != size)
>     {
>         return LFS_ERR_IO;
>     }
> 
>     return LFS_ERR_OK;
> }
> 
> // Erase a block. A block must be erased before being programmed.
> // The state of an erased block is undefined. Negative error codes
> // are propogated to the user.
> // May return LFS_ERR_CORRUPT if the block should be considered bad.
> static int _lfs_flash_erase(const struct lfs_config* c, lfs_block_t block)
> {
>     struct rt_mtd_nor_device* mtd_nor;
> 
>     RT_ASSERT(c != RT_NULL);
>     RT_ASSERT(c->context != RT_NULL);
> 
>     mtd_nor = (struct rt_mtd_nor_device*)c->context;
>     if (rt_mtd_nor_erase_block(mtd_nor, block * c->block_size, c->block_size) != RT_EOK)
>     {
>         return LFS_ERR_IO;
>     }
> 
>     return LFS_ERR_OK;
> }
> In the interface, the concept of a block is a sector, which is the minimum erasable unit of the W25Q128 chip, with a size of 4096. The write buffer of the W25Q128 is only 256 bytes.

Do you need more information for the analysis? If so, I'll do my best to provide it.

mc-xgl, 2025-06-03:
@geky Do you need more information for the analysis? If so, I'll do my best to provide it.

mc-xgl, 2025-06-03:
> [@geky](https://github.com/geky) Do you need more information for the analysis? If so, I'll do my best to provide it.

Compared to this issue, I would prefer if Littlefs could provide an interface to check for file system corruption. Currently, the LFS_ERR_CORRUPT problem can be randomly exposed through interfaces such as write, read, open, and close. I am unable to capture the error in advance and have no means of recovery. When the problem occurs, my only option is to completely erase all data and rebuild the file system, which is fatal for me. I cannot afford data loss or the impact of a "time-bomb" situation. Therefore, can Littlefs provide a way to check the file system so that I can perform a preliminary check before executing logic?

geky, 2025-06-03:
This configuration looks correct to me. It could always be a filesystem bug. Unfortunately, these things are complicated and require a lot of context to understand what is going wrong.

Usually the next step would be to reproduce locally, to completely rule out hardware and narrow in on what the bug could be. littlefs's test runner is a good target for this ([example](https://github.com/littlefs-project/littlefs/blob/master/tests/test_dirs.toml), [some instructions](https://github.com/littlefs-project/littlefs?tab=readme-ov-file#testing), sorry, it's not documented well at the moment). But it sounds like your application is quite involved which would make this difficult...

> Compared to this issue, I would prefer if Littlefs could provide an interface to check for file system corruption. Currently, the LFS_ERR_CORRUPT problem can be randomly exposed through interfaces [...]

littlefs v3 (https://github.com/littlefs-project/littlefs/pull/1111) may be interesting. It adds full [checksumming](https://github.com/littlefs-project/littlefs/pull/1111#n-gcksums) and a number of APIs to check on-disk errors early (`LFS3_M_CKMETA/CKDATA`, `lfs3_fs_ckmeta/ckdata`, etc). It's also better tested and intended to be the successor to littlefs v2.

However it's still a work in progress and will take some time to stabilize.

mc-xgl, 2025-06-04:
@geky Thanks for recommending littlefs V3.I'm quite interested in it, but now I'm focusing on the project issue. During recent tests, I found the problem occurs frequently in a demo, almost every run. Here's the demo:

```
// LFS file read/write test

uint8_t  read_buff[128*1024];
uint8_t  write_buff[128*1024];

uint8_t  oflc_path[128];
    // Write file

    for(int i = 0; i < sizeof(write_buff); i++)
    {
        write_buff[i] = i;
    }
    rt_kprintf("LFS data read start\r\n");
    for(int i = 0; i < 128; i += 1)
    {
        sprintf((char*)oflc_path,"/olcache/tf/0/%d/dfile/0/0",i);
    
        int fd = open((char*)oflc_path, O_RDONLY);
        if (fd < 0)
        {
            rt_kprintf("\r\n[err]: open [%s] failed\r\n",oflc_path);
            break;
        }
        
        if (read(fd, read_buff, sizeof(read_buff)) < sizeof(read_buff))
        {
            rt_kprintf("\r\n[err]: read [%s] failed\r\n",oflc_path);
            close(fd);
            break;
        }
        
        close(fd);
        
        if(memcmp(read_buff,write_buff,sizeof(read_buff)) != 0)
        {
            rt_kprintf("\r\n[err]: data cmp failed at [%s]\r\n",oflc_path);
            err_cnt ++;
            break;
        }
        else {
            rt_kprintf("\rdata cmp ok:[%d:%d]",i,128);
        }
        
    }
    
    rt_kprintf("LFS data write start\r\n");
    
    for(int i = 0; i < 128; i += 1)
    {
        sprintf((char*)oflc_path,"/olcache/tf/0/%d/dfile/0/0",i);
        
        int fd = open((char*)oflc_path, O_RDWR | O_CREAT | O_TRUNC);
        if (fd < 0)
        {
            rt_kprintf("\r\n[err]: open [%s] failed\r\n",oflc_path);
            goto ERR_HANDLE;
        }
        
        if (write(fd,write_buff , sizeof(write_buff)) < sizeof(write_buff))
        {
            rt_kprintf("\r\n[err]: write [%s] failed\r\n",oflc_path);
            close(fd);
            goto ERR_HANDLE;
        }
        
        if (-1 == lseek(fd, 0, SEEK_SET))
        {
            rt_kprintf("\r\n[err]: lseek [%s] failed\r\n",oflc_path);
            close(fd);
            goto ERR_HANDLE;
        }
        
        if (read(fd, read_buff, sizeof(read_buff)) < sizeof(read_buff))
        {
            rt_kprintf("\r\n[err]: read [%s] failed\r\n",oflc_path);
            close(fd);
            goto ERR_HANDLE;
        }
        
        close(fd);
        
        if(memcmp(read_buff,write_buff,sizeof(read_buff)) != 0)
        {
            rt_kprintf("\r\n[err]: data cmp failed at [%s]\r\n",oflc_path);
            err_cnt ++;
            break;
        }
        else {
            rt_kprintf("\rdata cmp ok:[%d:%d]",i,128);
        }
        
    }
    rt_kprintf("DFS restart\r\n");
```

The demo log is as follows:

```
LFS data read start

[err]: open [/olcache/tf/0/0/dfile/0/0] failed
LFS data write start
data cmp ok:[53:128]
[err]: read [/olcache/tf/0/54/dfile/0/0] failed

[err]: err handler runing
dfs restart
[00226.030]<6>[DEVICE][app/misc/device/device.c][263]:Ready reset!!!
```

Once the issue occurs, the only way to fix it is by erasing the FLASH.Could you help me analyze the cause?

mc-xgl, 2025-06-04:
So far, SFUD and FAL have been ruled out as the cause of the problem. The current investigation is focused on DFS, Littlefs, and RTT's MTD driver.