littlefs-project/littlefs #889: How do I truncate size file with last data (beginning file) was removed?

votuananhs, 2023-11-08:
Hello all,
I have logging data file. And I would like to remove in last data in case size file is going to limit size for that. I found lfs_file_truncate() function. But I have tried to run sample code with this function. It will keep data and size after running this command from beginning to size requested. That it doesn't meet my expectation. Do we have any solution for that with littlefs ? Many thanks in advance for your support/answer !

geky, 2023-11-14:
Hi @votuananhs, unfortunately this is not provided by littlelfs's APIs, and is in general not an operation filesystems support.

The best solution for logging over a filesystem seems to currently be maintaining multiple copies of the log file: https://github.com/littlefs-project/littlefs/issues/104

Though users note that multiple small files may be a better solution: https://github.com/littlefs-project/littlefs/issues/862

geky, 2023-11-14:
Though if it's any consolation, I'm currently looking into an API that would provide exactly this. It's very experimental though.