littlefs-project/littlefs #890: assertion "lfs->cfg->cache_size % lfs->cfg->read_size == 0" in function lfs_init

zhujhhh, 2023-11-09:
assertion "lfs->cfg->cache_size % lfs->cfg->read_size == 0" in function lfs_init.  The value of cache_size at the assertion is 2048 and the value of read_size is 2147672080. The value of read_size is obviously wrong. Has anyone experienced this problem?

geky, 2023-11-14:
Hi @zhujhhh, littlefs doesn't modify `lfs->cfg->read_size`, so I have to assume the issue is above littlefs.

Do you use littlefs as a part of another library? If so you may need to raise an issue there for uninitialized configuration of littlefs.

If you are declaring `struct lfs_config` yourself, are you not setting `read_size`?  `read_size=1` is valid and common for NOR flash devices.

Best-Hanc, 2025-03-21:
https://github.com/littlefs-project/littlefs/issues/1042