littlefs-project/littlefs #907: "Discussions" should be enabled for this repository

andriyndev, 2023-12-14:
Currently, all the discussions and questions are contained in the "Issues" section while they are more suitable for the "Discussions" section which is currently disabled for this repository. In my opinion, the "Discussions" section should be enabled for better organization which will allow, in particular, to simplify search of already created issues and discussions (questions).

geky, 2023-12-19:
I'm open to feedback on this, but I'm hesitant to adopt GitHub discussions. In part because once it's enabled, it would be destructive to disable it if it added no value.

This repo predates GitHub discussions, so issues were the only option at the time. Now that there has quite a bit of valuable discussion and feedback, you would still want to search both the issue tracker and discussions for relevant info. I think splitting things up risks making it harder to find shared questions than it already is.

Also I don't think the delineation of question/discussion/issue is quite so clear. Many times an innocent question turns out to be a real issue, and an honest issue turns out to be a configuration mistake. Last I checked GitHub does not allow converting a discussion to an issue or vice-versa.

I'd be curious if I'm missing something significant though.

dpgeorge, 2023-12-19:
We use Discussions for MicroPython and it works pretty well for us: https://github.com/orgs/micropython/discussions

> Also I don't think the delineation of question/discussion/issue is quite so clear.

For us it's a pretty clear line: is this something that needs to be fixed/improved/added in the main repo?  If yes it's an issue to track progress of that.  If no then it's a discussion topic.

> Last I checked GitHub does not allow converting a discussion to an issue or vice-versa.

You can definitely convert issues to/from discussion threads.  That's quite a useful feature.

geky, 2023-12-19:
> You can definitely convert issues to/from discussion threads. That's quite a useful feature.

Oh! That's good to hear. I'll have to play around with it.

Though this is probably low priority given other things. Apparently there's also a way to bulk migrate issues by label, which is awfully convenient.

geky, 2023-12-20:
It doesn't seem like you can convert discussion to threads. Conversion is one-way, with just a link to create a template issue pointing to the discussion. Though I think I can see how this flow could work.

geky, 2023-12-20:
The polls/voting systems do seem like great ways to aggregate feedback from users though