littlefs-project/littlefs #908: Unexpected data size information loss (zero)

sengulhamza, 2023-12-14:
Hello,

In our R&D project, we are using nRF9160 and mx25r64 external flash.
The project uses Zephyr as an OS. 
Nordic SDK 2.4.0
v2.4.0/modules/littlefs git commit hash _ca583fd297ceb48bced3c2548600dc615d67af24_
We are running LittleFS to use mx25r64.
We did our tests all day long, there was no problem, but we saw a problem in two of our devices.
The file system is mounting, we can list and see the files, but the file size returns zero as the value in the reading stage. It seems like the file is corrupted. The file is there, but the size is 0.
This problem was seen in our devices that have finished their software and are working in the field.

```
/**
 * The function checks if a file exists at the specified path and returns the appropriate status.
 *
 * @param path A string representing the file path to check for existence.
 *
 * @return an app_status_t enum value. It can return either APP_OK or APP_ERR, depending on whether the
 * file exists or not.
 */
app_status_t file_mngr_check_file_exist(const char *path, size_t *file_size)
{
    if ( !path ) {
        LOG_ERR("path is NULL to write");
        return APP_ERR;
    }
    char fpath[FILE_MNGR_MAX_PATH_LEN];
    snprintf(fpath, sizeof(fpath), "%s/%s", s_mp->mnt_point, path);
    struct fs_dirent entry;
    int rc = fs_stat(fpath, &entry);
    if (!rc) {
        if (entry.type == FS_DIR_ENTRY_FILE) {
            if (file_size) {
                *file_size = entry.size;
            }
            LOG_INF("%s is exist. Size is %d", path, entry.size);
        }
        return APP_OK; //File is exit.
    }
    LOG_ERR("%s is not exist.", path);
    return APP_ERR; //File is not exit.
}

```

**entry.size returns 0.**

When I delete and rewrite the file, or overwrite it, and read it again, there is no problem, but it contains important calibration data. I need to remove the device from the field.

What could be the cause, I am open to your suggestions.

Sincerely,

geky, 2023-12-19:
Hi @sengulhamza,

Is it possible you're losing power between creating a new file and the first call to `lfs_file_sync`/`lfs_file_close`?

When you create a new file with `lfs_file_open` it creates a zero-length file to put the filename somewhere. If you lose power before `lfs_file_sync`/`lfs_file_close` the zero-length file will still be there.

There some work planned to avoid this case (basically an "ignore me" file type), but it is not yet implemented.

sengulhamza, 2023-12-25:
Maybe, I am losing the power, I am not sure actually. We are working on new HW design of device. When it ready I will try on it. For now, I am saving critical parameters to internal flash too. 

Thank you for reply. @geky