littlefs-project/littlefs #922: Read and write data issue.

TriTerraIT, 2024-01-17:
I am using LittleFS in esp32 module. I have to write some uint32_t data into the file. I have write 266 and 522 (both in uint32_t) into the file. However, it become 10 when I read them again. 266mod256 and 522mod256 equal 10. However, I don't know why this issue happen. The other number from 10-600 is completely fine. Is someone have this experience before? Thank you

I have write a number into the LittleFS file but it change after I read it. This issue can be reproduced everytime.

M-Bab, 2024-07-03:
Can you send a code snippet where you do the `fwrite` command writing the numbers?

geky, 2024-09-21:
Sounds like you might be calling write with `size=1`, when you should use `size=sizeof(uint32_t)`.

But seconding @M-Bab, it's hard to know without the actual code.