littlefs-project/littlefs #932: user_provided_block_device_prog

JohannesLichtenberger, 2024-01-31:
Does this function have to write a single page/block or does it have to account for writing multiple pages? I think, regarding the source only a single block gets flushed everytime, so that might be sufficient already!? :-)

kind regards
Johannes

geky, 2024-02-02:
Hi @JohannesLichtenberger, the bd's `read` and `prog` functions don't cross `block_size` boundaries, but they can be called with requests that are multiples of the `read_size` and `prog_size`.

`prog_size` is _sometimes_ equivalent to "page" on some devices, sometimes not. Unfortunately the terminology gets a bit confusing and conflicting when you look at storage across different technologies...

JohannesLichtenberger, 2024-02-05:
Oh ok, so I'll have to use a function, which can write several pages to a Winbond W25Q :-)

geky, 2024-02-05:
The benefit of this approach is sometimes hardware can support faster writes over larger prog sizes. This wouldn't be possible if littlefs broke up the progs for you.

> Winbond W25Q

I believe this chip supports byte-level progs. You could set `prog_size=1` and then opportunistically use larger prog operations in your prog function. This would reduce the overhead of metadata padding.

But that depends on how much effort you want to put in. There's nothing functionally wrong with programming a full page every time.

closed 2024-03-06 by JohannesLichtenberger