littlefs-project/littlefs #933: lfs_file_write() return a error message of LFS_ERR_NOENT

filkli, 2024-01-31:
hello

I am using the littlefs 0f 2.8.01，it working normally for a long time,suddenly,lfs_file_write() return a error message(LFS_ERR_NOENT).Only formatting can restore normal, and restarting will not solve the problem.
I am a chinese,my english is so bad,but wo want to tell you the question,i hope to get help from you.
thanks

evaneight, 2024-02-01:
I am seeing the same issue with lfs 2.8, disc v.2.1
Block 0 and block 1 seem to be corrupted after extensive use.
Is it possible a recent change introduced this problem, is there an "older known good version" that anyone can recommend.

filkli, 2024-02-02:
> I am seeing the same issue with lfs 2.8, disc v.2.1 Block 0 and block 1 seem to be corrupted after extensive use. Is it possible a recent change introduced this problem, is there an "older known good version" that anyone can recommend.

thanks ,i am very happy to receive your reply.it is useful to me。I am using a newer version of v2.8.2.I hope it can avoid the question. if you alerady solved the question,please tell me.thanks .

Best wish to you !

evaneight, 2024-02-02:
No, I haven't resolved, I continue to test the flash drivers etc. which all seem fine. The test harness is pretty intensive and significant tests pass, but eventually it fails.

filkli, 2024-02-03:
> No, I haven't resolved, I continue to test the flash drivers etc. which all seem fine. The test harness is pretty intensive and significant tests pass, but eventually it fails.

I have been test the question for a long time,but it only appear a time。I think the version of V2.8.1 exist a bug possibly，But i have no valid evidence. I hope the author of littlefs can help me solve the question。Otherwise,i dare not to use the code .I want to commit  spifs to you,it is another file system,it is very stable。 Moreover,if you have other method or idea,please tell me.You can contace me quickly by means of  the emaile of 2460070599@qq.com,when it is convenient for you.
Best wishes to you!

evaneight, 2024-02-04:
hi

thanks for email

i am suspicious its a problem with littlefs, but need to do more checks..

i am noticing littlefs does different things some times and i assume this is because something is wrong in my code,

it could be hardware, power for example, but i still suspect software...

 

what flash chip are you using?

 

Sent: Saturday, February 03, 2024 at 4:42 AM

 

I have been test the question for a long time,but it only appear a time。I think the version of V2.8.1 exist a bug possibly，But i have no valid evidence. I hope the author of littlefs can help me solve the question。Otherwise,i dare not to use the code .I want to commit spifs to you,it is another file system,it is very stable。 Moreover,if you have other method or idea,please tell me.You can contace me quickly by means of the emaile of ***@***.***,when it is convenient for you.
Best wishes to you!

filkli, 2024-02-22:
> Sorry, I don't think I have much to add.
> 
> The last significant change was the introduction of the FCRC in [v2.6.0](https://github.com/littlefs-project/littlefs/releases/tag/v2.6.0) (so [v2.5.1](https://github.com/littlefs-project/littlefs/releases/tag/v2.5.1) may be interesting). Since then we've only had minor changes (additive features, heuristic tweaks, documentation, etc; [releases](https://github.com/littlefs-project/littlefs/releases)).
> 
> I don't know why `lfs_file_write` would return LFS_ERR_NOENT. This could happen if the file's metadata entry disappeared, but that shouldn't happen normally. It would be very interesting to know if this changed across a version bump.
> 
> > I want to commit spifs to you,it is another file system,it is very stable
> 
> spiffs is a good filesystem. It predates littlefs and takes a very different approach, so I wouldn't be surprised if there are cases where spiffs behaves better than littlefs.

thanks ,i am very happy to receive your reply.it is useful to me。if you have other,please tell me.

filkli, 2024-02-22:
> Running with my test harness... 2.0, 2.4, and 2.8 fail the same way (after some time) - with a file creation error. 2.1.1 gets through twice as many tests as the versions above, but ultimately fails with the same error.

Sorry,About the question,I have no good idea to solve it.I am ready  to use the old version of littlefs.(2.5.1)