littlefs-project/littlefs #935: Zephyr OS little FS file system Write timeing issue and read back issue after long hr testing

dhruvshah1997, 2024-02-01:
Hi 
I am working on one medical product using MT29F8G01ADBFD12 NAND flash(1 GB). I am using NRF52840 MCU and used Zephyr RTOS. I have ported a little FS file system on top of NAND flash. I can mount the file system, and create a file to the file system.

Now my requirement is to ECG data into Flash. Here ECG data is coming at a 10 ms rate. There is another data called IMU data which is coming at a rate of 5 ms—and temperature data at a rate of 1 second.

I have created 3 files, one for ECG, one for IMU and one for temperature. And append all data into respective files. I have taken 3 separate buffers to store 4096 bytes of data in the file system. So basically I have buffered 4096 bytes of data. Additionally, I am using a queue of 5 chunks(5 *4096 bytes). But still, some of the IMU and ECG data are missing, which means the file system append call will take a lot of time.

Please find my little FS file configuration:
const struct lfs_config cfg = {

	.context = NULL,
	.read = sh_lfsread,
	.prog = sh_lfsprog,
	.erase = sh_lfserase,
	.sync = sh_lfssync,

#ifdef LFS_THREADSAFE
	.lock = sh_lfslock,
	.unlock = sh_lfsunlock,
#endif

	.read_size = 4096,
	.prog_size = 4096,
	.block_size = 0x40000,
	.block_count = 4096,
	.block_cycles = 500, //500
	.cache_size = 4096,
	.lookahead_size = 4096,

};

I can append data for 1 day but there is some issue. I have added read logic. This means whatever I have appended to the file, I am reading back those data from the file. So if I append 30K records then I got a mismatch for 400-500 records.
Please find my code:
Here size is 4096 bytes.

#define MAX_READ_CHUNK_SIZE 2048
/**
 * @brief add data to file / create the file if not exist.
 * @arg filename - name of file
 * @return 0 if successful, negative value on error.
 */
int sh_littlefs::addtofile(char *filename, char *datatowrite, size_t size)
{
	int ret = 0;
	static uint32_t error_cnt = 0;
	struct lfs_info file_info;
	static uint8_t flag_test = 0;
	char buf[MAX_READ_CHUNK_SIZE];
	uint16_t id = size;
	uint8_t number_of_block_read = 0;

	if( (filename == NULL) || (datatowrite == NULL) || (size == 0) )
	{
		return -EINVAL;
	}

	if (mountstate != sh_littlefs::MOUNTED)
	{
		return -ENODEV;
	}

	//Now open the file in append mode
	ret = lfs_file_open(&lfs, &file, filename,  \
	LFS_O_RDWR | LFS_O_CREAT | LFS_O_APPEND);
	if (ret < 0)
	{
		LOG_ERR("FAIL: open %s: %d", filename, ret);
		return ret;
	}

	//Write the data to file
	ret = lfs_file_write(&lfs, &file, datatowrite, size);
	if (ret < 0)
	{
		LOG_ERR("FAIL: write %s: %d", filename, ret);
		return ret;
	}

	LOG_ERR("Size is %d", size);

	//Now move file pointer backwark for last X bytes
	ret = lfs_file_seek(&lfs, &file, -size, LFS_SEEK_END);
	if (ret < 0)
	{
		LOG_ERR("FAIL: read back file pointer %s: %d", filename, ret);
		return ret;
	}

	do
	{
		//Clear the buffer
		memset(buf, 0, MAX_READ_CHUNK_SIZE);

		//Now read first 2048 bytes from file
		ret = lfs_file_read(&lfs, &file, buf, MAX_READ_CHUNK_SIZE);
		if (ret < 0)
		{
			LOG_ERR("FAIL: read back %s: %d", filename, ret);
			return ret;
		}
		
		//Now compare the buffer
		if(strncmp((datatowrite+(number_of_block_read * MAX_READ_CHUNK_SIZE)), buf, MAX_READ_CHUNK_SIZE))
		{
			LOG_ERR("**[%d] : 2048 bytes not matching: %d**", number_of_block_read, ++error_cnt);
			//return -1;
		}
		id = id - MAX_READ_CHUNK_SIZE;
		number_of_block_read++;
	}while(id > 0);

	//Close the file
	ret = lfs_file_close(&lfs, &file);
	if (ret < 0)
	{
		LOG_ERR("FAIL: closing file %s: [rd:%d]", filename, ret);
		return ret;
	}

	return ret;
}

geky, 2024-02-02:
Hi @dhruvshah1997, thanks for opening an issue.

littlefs currently has some severe scalability issues, especially on "large" storage such as NAND flash. There are several open issues, but there are multiple contributing factors:
https://github.com/littlefs-project/littlefs/issues?q=is%3Aopen+is%3Aissue+label%3Aperformance

This is being worked on, but it's not a simple thing to improve.

You may want to consider setting `metadata_max` to artificially limit how large metadata can get to workaround some of the scalability issues, but this trades off storage space:
https://github.com/littlefs-project/littlefs/blob/f53a0cc961a8acac85f868b431d2f3e58e447ba3/lfs.h#L269-L273

But even then the throughput you're looking for may not be reachable with littlefs's current design.

CSC-Sendance, 2024-06-10:
Hi,

Do you have more details on this issue? We have a similar need but use a Winbond W25N01G NAND flash via SPI. It has a prog / page size of 2048 byte  with 128 kB erases.  We did some benchmarks on appending a single file with 300 byte file-write calls for 5k, respectively 100k times, and 400k times (which amounts to nearly the entire 1Gbit chip) - the latter is still in progress (I'll post it as soon as I have the data).

It seems with increasing occupation (or file size?) higher (file-)write times get more frequent While the vast MAJORITY is in the ~ 7ms range for the 2k byte page/prog size. 

LittleFS Parameters --> `no metadata_max limit`!:
```
const struct lfs_config cfg = {
    // block device operations
....
    // block device configuration
    .read_size = 2048,
    .prog_size = 2048,
    .block_size =  2048 * 64,
    .block_count = 1024,
    .block_cycles = 500,
    .cache_size = 2048, // "multiple of read and prog size, and a FACTOR of of block size"
    .lookahead_size = 2048,
    .compact_thresh = 0,
    .read_buffer = NULL,
    .prog_buffer = NULL,
    .lookahead_buffer = NULL,
    .name_max = 0,
    .file_max = 0,
    .attr_max = 0,
    .metadata_max = 0,
    .inline_max = 0};
```
 
<details><summary>Detailed NAND (LFS) benchmark without ECC, 5000x 300-byte buffer writes</summary>
<p>

[image]

```
Min: 0.0
Max: 25.39
Mean: 1.0249369873974794
Std: 2.4980383884536765
95% CI: [0.9556655405754659 , 1.0942084342194929]
```

Outlier analysis over 10ms per write call:

write call index to duration (ms):
```
3 -> 25.39
873 -> 11.72
1747 -> 14.65
2621 -> 11.72
3495 -> 17.58
4368 -> 11.72
```
[image]

</p>
</details> 

<details><summary>Detailed NAND (LFS) benchmark without ECC, 100k x 300-byte buffer writes</summary>
<p>

[image]

```
Total: 101115.77999996635
Min: 0.0
Max: 29.3
Mean: 1.0111679116791168
Std: 2.458219120299975
95% CI: [0.9959316549089265 , 1.026404168449307]
```

Outlier analysis over 10ms per write call:

write call index to duration (ms):
```
3 -> 25.39
873 -> 11.72
1747 -> 14.65
2621 -> 11.72
3495 -> 17.58
4368 -> 11.72
5242 -> 14.65
6116 -> 11.72
6990 -> 20.51
7863 -> 11.72
8737 -> 14.65
9611 -> 11.72
10485 -> 17.58
11358 -> 11.72
12232 -> 14.65
13106 -> 11.72
13980 -> 23.44
14853 -> 11.72
15727 -> 14.65
16601 -> 11.72
17475 -> 17.58
18349 -> 11.72
19222 -> 14.65
20096 -> 11.72
20970 -> 20.51
...
96987 -> 11.72
97861 -> 23.44
98734 -> 11.72
99608 -> 14.65
```

[image]

</p>
</details> 

It may be more insightful if I'd redo the tests with the page write size of 2kB  but 300bytes was closer to our use-case. We (currently) do no other buffering or queueing - I may do that later though. 

I suppose you can also see that the longer write times (>10ms) are relatively rare and follow a distinct pattern, so there is definately something that can be optimizable ;)

Getting back to `metadata_max` -  which we currently also make no use of - is there a particular downside to limit that to something like 2kB as mentioned?

geky, 2024-06-25:
Hi @CSC-Sendance, sorry for the late response.

My first thought was that you're now hitting lookahead buffer scans, but it's very interesting, and a bit wild, that you can clearly see the topology of the CTZ skip-list in your performance measurements.

It would be interesting to know the breakdown of read vs prog vs erase operations, this might be easy to measure with some simple global counters ([example](https://github.com/littlefs-project/littlefs/blob/d01280e64934a09ba16cac60cf9d3a37e228bb66/bd/lfs_emubd.c#L320)).

The best explanation I can think of is that this is simply the overhead necessary to build each [CTZ skip-list](https://github.com/littlefs-project/littlefs/blob/master/DESIGN.md#ctz-skip-lists) node.

The way our skip-lists work is every $n$ th block contains $\text{ctz}(n)+1$ pointers. This forms the spiky pattern you are seeing, with the tallest spikes being limited to $\log_2 n$.

What may be happening is building each of these pointers with a 2KiB reads ends up spending a lot of time reading unused data to find the relevant block address.

---

If this is the case I'm not sure there's an easy solution that doesn't require significant changes to littlefs. Multi-block caching would help avoid redundant reads, but brings its own complexity and has been low priority. At the very least B-trees would avoid this bottleneck, which is one thing I'm currently experimenting with as a CTZ skip-list replacement.

You could try _increasing_ the block size, since more blocks means fewer pointers. This would make metadata compaction performance worse, but that may just mean `metadata_max` is more useful.

CSC-Sendance, 2024-06-26:
Hi,

No problem! We have internally finished this "project" for now since we came to the conclusion that LittleFS with a NAND flash would be more suitable for our use case than our current NOR + FAT approach, despite the occurring limitations. However, it will not make it into active development until we have fixed a new major hardware revision.

> It would be interesting to know the breakdown of read vs prog vs erase operations, this might be easy to measure with some simple global counters ([example](https://github.com/littlefs-project/littlefs/blob/d01280e64934a09ba16cac60cf9d3a37e228bb66/bd/lfs_emubd.c#L320)).

I am still willing to help with this. If you wanna provide a modified LittleFS version that includes this (or a detailed description of what exactly you'd need) and the benchmark you'd like to receive, I can execute it for you.
 
> ...
> You could try increasing the block size, since more blocks means fewer pointers. This would make metadata compaction performance worse, but that may just mean `metadata_max` is more useful.

Interesting, but the maximum block size is defined by the flash itself (and its block erase operations etc.) or would you handle that more independently and just check for that in, e.g. the erase option and then perform, e.g., 2x erases if the "virtual" block size is doubled?

geky, 2024-06-26:
> Interesting, but the maximum block size is defined by the flash itself (and its block erase operations etc.) or would you handle that more independently and just check for that in, e.g. the erase option and then perform, e.g., 2x erases if the "virtual" block size is doubled?

Ah yeah. It's not well documented, but there's nothing stopping you from defining block_size as a multiple of the flash's actual erase size, and emulating the larger erase with multiple operations. This can be useful for working around issues with the block allocator scaling poorly with too many blocks, but usually also runs into issues with metadata compaction, metadata_max, etc.

I've been considering adding an additional `erase_size` config option, and doing this emulation littlefs-side, to make this easier for users, but haven't been sure if it's worth the code cost.

> I am still willing to help with this. If you wanna provide a modified LittleFS version that includes this (or a detailed description of what exactly you'd need) and the benchmark you'd like to receive, I can execute it for you.

Thanks! But it was more for curiosity's sake if it was little extra work, to confirm the above assumption.

Long term, the plan is to move away from CTZ skip-lists, so I don't think this will lead to anything actionable short-term.