littlefs-project/littlefs #943: data loss in littlefs

Aravind0812, 2024-02-14:
Hi, I am using w25Q512 flash memory. my cache size is 256 bytes, after filling 256 bytes littlefs writing to Flash is success, but it is not updating on metadata, when I close the file only is updating, otherwise is not updating on metadata. I didn't close the file after i open the file i lost my old data. my question is why is not updating on meta data (or) only file close or file sync only save the and update the value on metadata (or) any other method is available to save without call file-close or file-sync function.

geky, 2024-02-17:
Hi @Aravind0812, if I understand correctly, this is just how littlefs works.

In general, it is more efficient to wait as long as possible to update the metadata, because littlefs doesn't know how much data is being written until `lfs_file_close`.

`lfs_file_sync` allows you to manually update the metadata if you want to keep the file open. It's ok to call `lfs_file_sync` after every file write.

Aravind0812, 2024-02-18:
Thank you for the response, Now i understand.

closed 2024-02-18 by Aravind0812