littlefs-project/littlefs #946: lfs_filebd.c seems to assume errno values

yamt, 2024-02-27:
functions in lfs_filebd.c (eg. lfs_filebd_read) returns a negative errno. (eg. -errno)
it doesn't necessarily match with enum lfs_error values.

geky, 2024-02-28:
Hi @yamt, thanks for creating an issue.

Ah yes, I can see where this could be an problem.

The exact solution though seems a bit unclear. We really shouldn't be mapping all OS errors into littlefs errors in the bd since 1. littlefs doesn't define all OS errors and 2. this can mess up littlefs internals.

These should probably all be mapped to LFS_ERR_IO, though it's a shame to lose the extra error info. At least `lfs_filebd_create` can return some of the more useful errors, which is where we call `open`.

---

I'm curious, did you run into an actual error during use? From what I understand most modern OSs `read`/`write`/`lseek` don't really ever error unless you yank the harddrive out of the machine.

yamt, 2024-02-28:
> Hi @yamt, thanks for creating an issue.
>
> Ah yes, I can see where this could be an problem.
>
> The exact solution though seems a bit unclear. We really shouldn't be mapping all OS errors into littlefs [...]

i guess leaving the original errno in LFS_FILEBD_TRACE is good enough.
 
> I'm curious, did you run into an actual error during use? From what I understand most modern OSs `read`/`write`/`lseek` don't really ever error unless you yank the harddrive out of the machine.

no. i was just reading the code.