littlefs-project/littlefs #947: Provide a function for probing for LFS

rojer, 2024-02-27:
It would be nice to have a function that would probe a device for LFS and return the parameters if found (the `lfs_superblock_t` structure).

rojer, 2024-02-27:
naturally, we must at least know the block size.
here's my quick and dirty take: https://gist.github.com/rojer/cf1a99497c1f1e807fe0e44c059daa17
pulled out the relevant part of lfs_rawmount

geky, 2024-02-28:
I'm curious, what would this gain over `lfs_mount` + `lfs_fs_stat`?

Unless I'm missing something, everything that is in the superblock should be accessible via `lfs_fs_stat`.

rojer, 2024-02-28:
is lfs_mount guaranteed to not modify the fs?

the reason for probing is that we want to provide exact disk_version to lfs_mount to avoid upgrading the disk format (we want to preserve compatibility with older firmware for now).

rojer, 2024-02-28:
there is no runtime read-only mount option, only LFS_READONLY. i guess lfs_mount ro + lfs_fs_stat would be a way.
but also, probing, for example, should be quiet about finding gibberish, since finding something else is a valid expected result.
we also probe to determine which filesystem should be used - if device is upgrading from spiffs, we need to be able to detect it.
so overall, a separate probe api would be useful, i think,

geky, 2024-02-28:
> is lfs_mount guaranteed to not modify the fs?

Ah yes, `lfs_mount` involves no erase/prog operations.

In general littlefs has some pretty strong guarantees around not mutating the disk until an actual mutation API is called (`lfs_file_open`, `lfs_remove`, `lfs_fs_mkconsistent`, etc). Though I realize this isn't really documented.

> there is no runtime read-only mount option, only LFS_READONLY

I've been thinking of adding a runtime LFS_M_READONLY flag to `lfs_mount`, but it's low priority since the best API would require a breaking change, and it technically doesn't add any functionality given the above guarantees.

But there's other mount flags that could be useful in the future.

> but also, probing, for example, should be quiet about finding gibberish

This is a valid point. We should probably make a failing `lfs_mount` not log any errors, though this is a bit tricky in practice.

The notorious "Corrupted dir pair" log ([here](https://github.com/littlefs-project/littlefs/blob/f53a0cc961a8acac85f868b431d2f3e58e447ba3/lfs.c#L1366)) can probably be downgraded to LFS_DEBUG, though I'm not sure if there's other error logs that can be hit.

---

I do like the idea and design of the `lfs_probe` API, but by default I need to be against new APIs unless there are strong reasons for their inclusion, to combat API complexity, bug surface-area, maintenance overhead, etc.

Since this can be implemented on the user's side:

``` c
int lfs_probe(lfs_t *lfs, const struct lfs_config *cfg,
        struct lfs_fs_stat *fsstat) {
    int err = lfs_mount(lfs, cfg);
    if (err) {
        return err;
    }

    err = lfs_fs_stat(lfs, fsstat);
    if (err) {
        return err;
    }

    err = lfs_unmount(lfs);
    if (err) {
        return err;
    }

    return 0;
}
```

I'm also curious if there's any prior art. From what I understand most filesystems rely on `lfs_mount` failure for querying if a filesystem exists, for similar reasons to why there is no `fexists` like function`.

---

> the reason for probing is that we want to provide exact disk_version to lfs_mount to avoid upgrading the disk format (we want to preserve compatibility with older firmware for now).

I'm probably missing something, but would this be more easily accomplished by setting `cfg.disk_version = 0x00020000`?

If configured this way, littlefs should error with LFS_ERR_INVAL if it encounters an lfs2.1 disk version ([here](https://github.com/littlefs-project/littlefs/blob/f53a0cc961a8acac85f868b431d2f3e58e447ba3/lfs.c#L4442-L4443), oh look another error log)

rojer, 2024-02-28:
> setting cfg.disk_version = 0x00020000
yes, but that would hard-code the version and won't be as flexible during migration.
right now we mount with `cfg.disk_version = probed_version`.
you may or may not remember that we ship new filesystem images with our fw, that on its first boot mounts the old fs and copies over the volatile state. thus new code must support old disk format, obviously.
however, in order to support rollback, for the transitional period _old_ code must also be able to mount new fs: to perform reverse migration when old code boots with old initial fs image and mounts used-to-be-new fs to copy over the state.

thus our disk_ver migration plan is:

1) start shipping code that supports 2.1 but ship 2.0 filesystem images.
2) start shipping 2.1 images. at this point we won't be able to roll back anymore.

we implemented (1) about 6 months ago but haven't done (2) yet, so even our latest firmware still comes with disk_ver 2.0 images. when we update out image generation to 2.1 (probably in a month or so), we'll still maintain 6+ month rollback compatibility window because of the version probing behavior. had we hard-coded `cfg.disk_version = 0x00020000` in (1), we'd have a hard cut-off in terms of rollback since they'd refuse to mount 2.1 images.

geky, 2024-02-29:
Ah I see, thanks for clarifying.

Yeah, I can't think of a better solution with littlefs's current API.

If/when we have mount flags, adding LFS_M_NOUPGRADE (or more likely requiring LFS_M_UPGRADE to upgrade) would make a lot of sense.

rojer, 2024-02-29:
can't mount flags be added to `struct lfs_config`?

geky, 2024-02-29:
That's a good point, it would be inconsistent with `lfs_file_opencfg`, but there's nothing stopping us from adding flags to `struct lfs_config`.

I was also considering adding `lfs_mountcfg` (with flags) and deprecating `lfs_mount`, which would be a bit more disruptive but at least not an API breakage.

But there are some... more disruptive changes bubbling down the pipeline. If we're going to have major API breakages somewhat soon I think trying to add flags to `lfs_mount` is the way to go.

If things take longer than expected (as they usually do) I'll consider adding LFS_M_NOUPGRADE to `struct lfs_config`. It is a nice way to move things in the right direction.

geky, 2024-02-29:
> thus our disk_ver migration plan is:
> 
> 1) start shipping code that supports 2.1 but ship 2.0 filesystem images.
> 2) start shipping 2.1 images. at this point we won't be able to roll back anymore.

Just thinking out loud. Another option might be to try mounting first with `disk_version=0x00020000`, and if that fails try mounting with `disk_version=0x00020001`.

Not sure if this is better or worse than mount->fs_stat->unmount->mount.

rojer, 2024-02-29:
> trying to add flags to lfs_mount is the way to go.

but we'd need to store them anyway and presumably check? e.g. `LFS_M_READONLY` would need to be consulted on every mutating operation. might as well just add them to `struct lfs_config`. but anyway, up to you.

so, my wishlist for flags:
 - `LFS_M_READONLY` - extra safety when mounting old filesystem to migrate from, we don't want to mutate anything.
 - `LFS_M_NOUPGRADE` - we'd use this in general, to preserve whatever disk version the image was created with.
 - `LFS_M_SILENT` - silence errors, perhaps? when using mount to probe an unknown device, we don't want any screams in the logs.

geky, 2024-03-08:
> but we'd need to store them anyway and presumably check? e.g. `LFS_M_READONLY` would need to be consulted on every mutating operation. might as well just add them to `struct lfs_config`.

Ah sorry, this is in the context of some other changes that move away from needing `struct lfs_config` to be long-lived. The theory being cases where you can benefit from `struct lfs_config` being readonly are better handled with compile-time configuration. Otherwise `struct lfs_config` usually ends up in RAM anyways.

> `LFS_M_SILENT` - silence errors, perhaps? when using mount to probe an unknown device, we don't want any screams in the logs.

It's not very clear what the best solution is, but I think we may just want to treat errors during `lfs_mount` as unexceptional. Similar to `lfs_file_open`. So downgrade them to LFS_DEBUG.

rojer, 2024-12-23:
@geky ping on this one. i tried updating to v2.10.1 and it reminded me of the `lfs_probe` API i added for this purpose (finding out on-disk version w/o upgrading). also runtime read-only mode would be nice to have.

geky, 2025-01-29:
Hi @rojer, noted.

It may be interesting to note a runtime readonly flag is implemented on an experimental branch ([rbyd](https://github.com/littlefs-project/littlefs/tree/rbyd), [LFS_M_RDONLY](https://github.com/littlefs-project/littlefs/commit/0ab0406d537990e0bfa609df668e20035e19a2fd)), though I realize this doesn't really help you at the moment.

But my current and only priority is to get this experimental branch into a mergeable state. For better or worse, anything that doesn't contribute to this branch is going to be low-priority for a while, and, if I'm honest, I hope to put littlefs into a sort of soft "feature-freeze" until the rbyd work completes. Every new feature is something to backport, and it starts to get a bit counterproductive (though user feedback is valuable, so it's a complicated tradeoff).

---

As a short-term workaround, you could add checks that error in your bd prog/erase functions if you're in "probe mode".

I'd also welcome tests that assert `lfs_mount`+`lfs_fs_stat` does not prog/erase on it's own. This is an intentional design decision, though I realize the lack of guardrails may be a bit unnerving.

---

I vaguely remember changing the mount path to _not_ LFS_ERROR on an unformatted filesystem, but I think it was in the above experimental branch... So also not very useful...

---

This comment also prompted me to rethink the hypothetical `lfs_probe` function, and while I can see how it'd be convenient, I still think it's not entirely obvious what it would look like. Returning `lfs_fsinfo` would make sense, but `lfs_fs_size` may be unreasonably expensive for the user, and future features such as `lfs_fs_cksum` raise similar questions... Requiring `lfs_mount` avoids trying to oversimplify a complex API...

DvdGiessen, 2025-05-20:
> I'm also curious if there's any prior art. From what I understand most filesystems rely on lfs_mount failure for querying if a filesystem exists, for similar reasons to why there is no fexists like function`.

For reference, in MicroPython there is some code that tries to detect whether a block device contains a LittleFS filesystem: https://github.com/micropython/micropython/blob/186caf9f0326c9d61494a7d5c6d0408c0fef8485/extmod/vfs.c#L166-L206

In the Python interface for filesystems you can explicitly instantiate [a VFS class](https://docs.micropython.org/en/latest/library/vfs.html) for your desired filesystem and mount it:

```python
vfs.mount(vfs.VfsLfs2(blockdevice), "/")
```

But you can also just pass the block device directly to the `mount` function and it will try to automatically detect the filesystem:

```python
vfs.mount(blockdevice, "/")
```

Under the hood that uses the code linked above to determine which filesystem class should be used.

It also is not entirely correct; I found this issue because I went looking after it failed to detect my filesystem. Linking it here since it is an example of prior art and an actual use case.

geky, 2025-11-11:
> i just finished porting (copy-pasting) lfs_probe to 2.11 and wondering if it can be included in v3, in some form?

~~Probably not in the first version. As an API-only and non-breaking change (unlikely to affect disk format, no reason to piggyback on breaking changes), this would have low priority.~~

~~It's open for discussion later, but I'm still not sure what the value is over `mount -> stat -> unmount`.~~

Oh, I see where this might help _with_ v2 -> v3 migration.

But I think to be useful it v3 would need to support v2, which is currently (maybe controversially) not in scope.

With v3 not understanding v2, I think you'd still want your own probe wrapper:

``` c
uint32_t disk_version = 0;
struct lfs3_fsinfo lfs3info;
int err = lfs3_probe(&lfs3, &lfs3cfg, &lfs3info);
assert(!err || err == LFS3_ERR_CORRUPT);
if (!err) {
    disk_version = LFS3_DISK_VERSION;
} else {
    struct lfs2_fsinfo lfs2info;
    err = lfs2_probe(&lfs2, &lfs2cfg, &lfs2info);
    assert(!err || err == LFS3_ERR_CORRUPT);
    if (!err) {
        disk_version = lfs2info.disk_version;
    }
}
```

vs

``` c
uint32_t disk_version = 0;
int err = lfs3_mount(&lfs3, LFS3_M_RDONLY, &lfs3cfg);
assert(!err || err == LFS3_ERR_CORRUPT);
if (!err) {
    err = lfs3_unmount(&lfs3);
    assert(!err);
    disk_version = LFS3_DISK_VERSION;
} else {
    err = lfs2_mount(&lfs2, &lfs2cfg);
    assert(!err || err == LFS3_ERR_CORRUPT);
    if (!err) {
        struct lfs2_fsinfo lfs2info;
        err = lfs2_fs_stat(&lfs2, &lfs2info);
        assert(!err);
        disk_version = lfs2info.disk_version;
    }
    err = lfs2_unmount(&lfs2);
    assert(!err);
}
```

(Quick and a bit handwavey)

---

On the good news, littlefs3 at least supports LFS3_M_RDONLY. To make the read-only behavior explicit.

There's also an effort to move away from on-disk minor versions in littlefs3 and instead use compat-flags for new features. That and to not auto-upgrade. I think that was a mistake rooted in naivety...

In the context of https://github.com/littlefs-project/littlefs/issues/1167, I wonder if need for an `lfs_probe`-esque API is a symptom of the clunkiness introduced by auto-upgrading? If LFS_MULTIVERSION instead mounted whatever version it found and upgrading required an explicit operation, would `lfs_probe` still be necessary?

Also apologies if anyone here warned me against this exact thing. Feel free to tell me you told me so.

geky, 2025-11-11:
> If LFS_MULTIVERSION instead mounted whatever version it found and upgrading required an explicit operation, would `lfs_probe` still be necessary?

Actually, thinking about this, we could still add this as a new LFS_LAZYMULTIVERSION mode without breaking API compatibility. Though unsure it's worth it at this stage with littlefs3 on the way.