littlefs-project/littlefs #949: Hardfault in lfs_dir_relocatingcommit, LFS 2.9, LFS_ASSERT(pdir) with pdir null

HariTheCoder341, 2024-03-01:
We are experiencing hardfault time to time in this scenario (while(true) loop).
We have 3 directory created, "Records", "ToSend", "Archive"

1. creating a file in "Records" and adding data to it (random size)
2. renaming the file from "Records" and moving it to "ToSend"
3. Copying the content of "ToSend" file in a single file in "Archive", appending content to it

Hardfault happens in function "lfs_dir_relocatingcommit" line 2226 hitted by LFS_ASSERT(pdir); because pdir is 0 (null).
It happen here:

```
01/03/2024 10:03:01: ../Core/lfs/lfs.c:2438:debug: Relocating {0xa5d, 0x1613} -> {0xcd1, 0xa5d}
01/03/2024 10:03:01: ../Core/lfs/lfs.c:2535:debug: Fixing move while relocating {0xa28, 0x3de} 0x0
```

After reboot and during mount:
```
01/03/2024 10:03:12: ../Core/lfs/lfs.c:4534:debug: Found pending gstate 0xcff0000000000a28000003de
01/03/2024 10:03:12: ../Core/lfs/lfs.c:4883:debug: Fixing move {0xa28, 0x3de} 0x0
01/03/2024 10:03:12: ../Core/lfs/lfs.c:4960:debug: Fixing half-orphan {0xa5d, 0x1613} -> {0xcd1, 0xa5d}

```
We also noticed that lfs_dir_relocatingcommit is called with pdir = NULL in three points, line 2433, 2494, 2545. 
Is this intentional?

Why could this problem happen and how can we avoid this?

But, if we mount the fs at the beginning and unmount it at each while cycle, it does not happen.

Thank you

geky, 2024-03-08:
Hi @HariTheCoder341, thanks for creating an issue.

This is going to be difficult to debug. Is it possible to reduce this down into a locally-reproducible example? Preferably in a littlefs test case ([example](https://github.com/littlefs-project/littlefs/blob/f53a0cc961a8acac85f868b431d2f3e58e447ba3/tests/test_files.toml#L2-L24)).

---

> We also noticed that lfs_dir_relocatingcommit is called with pdir = NULL in three points, line 2433, 2494, 2545.

Are you sure you're using v2.9? I don't think these line numbers quite line up.

The `pdir` argument here is actually a side-effect of `lfs_dir_relocatingcommit`. It contains the previous mdir in the dir's linked-list if the mdir needs to be dropped (`mdir.count=0`, LFS_OK_DROPPED). We need to find the `pdir` in `lfs_dir_relocatingcommit` to determine how to update the mlist, but we can't actually drop in `lfs_dir_relocatingcommit` without recursion. A lot of the mess in these functions is tip-toeing around a flattened recursive algorithm.

But not every call to `lfs_dir_relocatingcommit` can drop. This would not be solved by always providing `pdir`, because the layer above would not be able to handle the resulting LFS_OK_DROPPED state correctly.

The `Fixing move while relocating` message ([here](https://github.com/littlefs-project/littlefs/blob/f53a0cc961a8acac85f868b431d2f3e58e447ba3/lfs.c#L2535)) is followed immediately by an `lfs_dir_relocatingcommit` call with a delete tag, which is probably resulting a drop, which littlefs doesn't expect.

HariTheCoder341, 2024-03-25:
Hi,
In the past few days I tried to write a test to reproduce the problem, without success.
The operations are the same as in the real application, but in a Linux environment they do not lead to the problem, which instead persists on STM32.
I attach the test code.

I confirm that by replacing the lfs_rename with a copy function, which replicates the purpose, we have no reset after extensive and continuous testing for 10 consecutive days.

I also generated a log file with LFS_YES_TRACE active, but it exceeds 1.5 GB, so I don't think it can be attached here or that it can be of any help, I'm only publishing the final part.

It takes a day and a half for the problem to occur... not knowing how to proceed, we will comment out the lfs_rename in favor of the copy for now.

[lfstest.zip](https://github.com/littlefs-project/littlefs/files/14742608/lfstest.zip)
[log_pub.zip](https://github.com/littlefs-project/littlefs/files/14742649/log_pub.zip)

kenchow96, 2026-10-24:
We hit this in a downstream firmware (MicroPython v1.26.1, which carries LittleFS v2.11.0, on an RP2040) and have a
small, deterministic reproduction on a RAM block device, in case it helps.

**Call path** (line numbers in v2.11.0's `lfs.c`; in v2.11.3 / master add 2, and 10 for `lfs_fs_pred`):

1. `lfs_rename` of a file out of a directory that holds only that entry (`e`) into another directory (`d`) whose
   commit relocates it (block_cycles reached), where `e` is `d`'s predecessor on the threaded list.
2. `lfs_dir_orphaningcommit` relocates `d` and fixes the pending move in the predecessor ("Fixing move while
   relocating", :2565-2573), calling `lfs_dir_relocatingcommit(lfs, &pdir, lpair, ..., NULL)` (:2580-2585), i.e.
   with `pdir == NULL`.
3. That commit deletes `e`'s last entry, so `lfs_dir_relocatingcommit` takes the drop branch
   `if (hasdelete && dir->count == 0)` (:2257), hits `LFS_ASSERT(pdir)` (:2258) and calls
   `lfs_fs_pred(lfs, dir->pair, pdir)` (:2259).
4. With asserts compiled out, `lfs_fs_pred` writes `pdir->tail[0] = 0` (:4798) through NULL.

**Reproduction** (asserts on; v2.11.0 fails `LFS_ASSERT(pdir)` at lfs.c:2258, master at :2260):

```c
// lfs_rename across directories reaches lfs_fs_pred with pdir == NULL (littlefs #949)
// cc -I. repro.c lfs.c lfs_util.c && ./a.out
#include "lfs.h"
#include <stdio.h>
#include <string.h>

#define BLOCK_SIZE  4096
#define BLOCK_COUNT 96
static uint8_t disk[BLOCK_COUNT][BLOCK_SIZE];

static int bd_read(const struct lfs_config *c, lfs_block_t b, lfs_off_t off,
        void *buf, lfs_size_t n) {
    memcpy(buf, &disk[b][off], n); return 0;
}
static int bd_prog(const struct lfs_config *c, lfs_block_t b, lfs_off_t off,
        const void *buf, lfs_size_t n) {
    memcpy(&disk[b][off], buf, n); return 0;
}
static int bd_erase(const struct lfs_config *c, lfs_block_t b) {
    memset(disk[b], 0xff, BLOCK_SIZE); return 0;
}
static int bd_sync(const struct lfs_config *c) { return 0; }

static const struct lfs_config cfg = {
    .read = bd_read, .prog = bd_prog, .erase = bd_erase, .sync = bd_sync,
    .read_size = 32, .prog_size = 256, .block_size = BLOCK_SIZE,
    .block_count = BLOCK_COUNT, .cache_size = 1024, .lookahead_size = 32,
    .block_cycles = 1,      // relocate a metadata pair at every compaction
};

static void put(lfs_t *lfs, const char *path, lfs_size_t size) {
    static uint8_t buf[256];
    lfs_file_t f;
    memset(buf, 'a', sizeof buf);
    lfs_file_open(lfs, &f, path, LFS_O_WRONLY | LFS_O_CREAT | LFS_O_TRUNC);
    lfs_file_write(lfs, &f, buf, size);
    lfs_file_close(lfs, &f);
}

int main(void) {
    lfs_t lfs;
    memset(disk, 0xff, sizeof disk);
    lfs_format(&lfs, &cfg);
    lfs_mount(&lfs, &cfg);
    lfs_mkdir(&lfs, "d");
    lfs_mkdir(&lfs, "e");
    put(&lfs, "e/x", 30);                   // e holds a single entry
    char name[8];
    for (int i = 0; i < 8; i++) {
        snprintf(name, sizeof name, "d/f%d", i);
        put(&lfs, name, 100);
    }
[... 21 lines cut ...]
-    if (hasdelete && dir->count == 0) {
+    if (hasdelete && dir->count == 0 && pdir) {
         LFS_ASSERT(pdir);
```

As noted earlier in this thread, the callers that pass NULL cannot handle a drop. With this change the emptied pair is
committed and kept instead. When the emptied pair is a directory's head (its predecessor not split), as in the
reproduction, that is exactly what the code does when it is given a `pdir`: `lfs_fs_pred` runs only to look at
`pdir->split`, and nothing is dropped. Only an emptied continuation of a split directory would be kept where a drop
was intended. In an instrumented build, every NULL-`pdir` occurrence we saw (64, in random workloads) had an unsplit
predecessor; in 180 further runs with split directories we never reached it.

**What we tested** (host builds of v2.11.0 with the change; we have not run littlefs's own test suite):
- the reproduction above, and the same with 0-200 rewrites of `d/z`: stock asserts at 11, 50, 89, 128 and 167
  rewrites; with the change every rename returns 0, `d/x` exists, `e/x` is gone, and a remount plus
  `lfs_fs_mkconsistent` return 0. The same on master;
- the tmp -> data loop above, 10 000 renames at 30 B and at 600 B files on 2 816 blocks of 4 KiB (block_cycles 100;
  files opened as MicroPython opens them, with an 8-byte custom attribute):
  stock reaches the NULL write at rename 1 600; with the change all 10 000 return 0, the filesystem checks
  consistent on a second mount (`lfs_fs_mkconsistent`), and further writes after a remount succeed;
- 80 randomised runs (1 500 operations each: whole-file writes, renames across three directories, removes; block_cycles
  1 and 3): stock reaches the NULL write in 49; with the change every call succeeds and the final listing matches a
  model of the files; in the 31 runs stock survives, both builds give byte-identical results;
- about 5 000 scenario runs from our own test corpus, with every program and erase traced and a block-device error
  injected at each write in turn: byte-identical to stock (none of them reaches this path).

HariTheCoder341, 2026-10-27:
Thanks for the info.
I reactivated the board test to see if it still happens, unfortunately the project context has changed a bit, as it has evolved over the years. 
If I can replicate the problem, I will install your patch to verify the outcome. 
I'll keep you informed, it's running...

HariTheCoder341, 2026-10-28:
Thanks @kenchow96 for the analysis and the reproduction, I can confirm both
the bug and the fix.

**Results**
- your reproduction fails on v2.9.0, v2.9.3 and v2.11.3, and passes with the patch
- I turned it into a test case for the suite (below): 50/640 permutations fail
  without the fix (default, eeprom and nand geometries), 640/640 pass with it
- full suite with the fix: 12406/12406 passed, 28741 power-losses (Linux and Cygwin)
- on our STM32 board with `block_cycles = 2`, the original crash reappears
  within minutes without the fix; with it, the board ran ~15.5 h continuously
  (>700 MB written and archived) with no failures

The failing test log shows the same "Fixing move while relocating" sequence as
my original report, so this is the same bug.

Since the fix is yours, would you like to open the PR? Feel free to include
the test as-is (it fits in `tests/test_relocations.toml`). One open point for
review: if the emptied mdir is the continuation of a split directory, it is
now kept (empty) instead of dropped, and the test doesn't cover that case.

<details>
<summary>test_relocations_rename_empties_pred</summary>

```toml
# rename out of a directory that ends up empty, into a directory whose commit
# relocates it, where the source dir is the destination's predecessor in the
# metadata list. "Fixing move while relocating" then commits the delete with
# pdir == NULL, which used to reach the drop branch and lfs_fs_pred(..., NULL)
# (https://github.com/littlefs-project/littlefs/issues/949)
[cases.test_relocations_rename_empties_pred]
defines.REWRITES = 'range(0, 64)'
defines.BLOCK_CYCLES = [1, 2]
code = '''
    lfs_t lfs;
    lfs_format(&lfs, cfg) => 0;
    lfs_mount(&lfs, cfg) => 0;
    // created in this order, "e" ends up as "d"'s predecessor
    lfs_mkdir(&lfs, "d") => 0;
    lfs_mkdir(&lfs, "e") => 0;

    uint8_t buffer[128];
    lfs_file_t file;
    // "e" holds a single entry
    memset(buffer, 'x', sizeof(buffer));
    lfs_file_open(&lfs, &file, "e/x",
            LFS_O_WRONLY | LFS_O_CREAT | LFS_O_TRUNC) => 0;
    lfs_file_write(&lfs, &file, buffer, 30) => 30;
    lfs_file_close(&lfs, &file) => 0;

    // fill "d" a bit
    for (unsigned i = 0; i < 8; i++) {
        char path[16];
        sprintf(path, "d/f%u", i);
        memset(buffer, 'a'+i, sizeof(buffer));
        lfs_file_open(&lfs, &file, path,
                LFS_O_WRONLY | LFS_O_CREAT | LFS_O_TRUNC) => 0;
        lfs_file_write(&lfs, &file, buffer, 100) => 100;
        lfs_file_close(&lfs, &file) => 0;
    }

    // rewrite a file in "d" to bring it closer to a compaction/relocation
    memset(buffer, 'z', sizeof(buffer));
    for (unsigned i = 0; i < REWRITES; i++) {
        lfs_file_open(&lfs, &file, "d/z",
                LFS_O_WRONLY | LFS_O_CREAT | LFS_O_TRUNC) => 0;
        lfs_file_write(&lfs, &file, buffer, 40) => 40;
        lfs_file_close(&lfs, &file) => 0;
    }

    // move the only entry of "e" into "d"
    lfs_rename(&lfs, "e/x", "d/x") => 0;
    lfs_unmount(&lfs) => 0;

    // check everything after a remount
    lfs_mount(&lfs, cfg) => 0;
    lfs_fs_mkconsistent(&lfs) => 0;
    struct lfs_info info;
    lfs_stat(&lfs, "e/x", &info) => LFS_ERR_NOENT;
    lfs_stat(&lfs, "d/x", &info) => 0;
    assert(info.type == LFS_TYPE_REG);
    assert(info.size == 30);

    uint8_t rbuffer[128];
    lfs_file_open(&lfs, &file, "d/x", LFS_O_RDONLY) => 0;
    lfs_file_read(&lfs, &file, rbuffer, sizeof(rbuffer)) => 30;
    lfs_file_close(&lfs, &file) => 0;
    memset(buffer, 'x', sizeof(buffer));
    memcmp(rbuffer, buffer, 30) => 0;

    for (unsigned i = 0; i < 8; i++) {
        char path[16];
        sprintf(path, "d/f%u", i);
        lfs_file_open(&lfs, &file, path, LFS_O_RDONLY) => 0;
        lfs_file_read(&lfs, &file, rbuffer, sizeof(rbuffer)) => 100;
        lfs_file_close(&lfs, &file) => 0;
        memset(buffer, 'a'+i, sizeof(buffer));
        memcmp(rbuffer, buffer, 100) => 0;
    }

    // "e" is empty and can be removed
    lfs_dir_t dir;
    lfs_dir_open(&lfs, &dir, "e") => 0;
    lfs_dir_read(&lfs, &dir, &info) => 1;
    lfs_dir_read(&lfs, &dir, &info) => 1;
    lfs_dir_read(&lfs, &dir, &info) => 0;
    lfs_dir_close(&lfs, &dir) => 0;
    lfs_remove(&lfs, "e") => 0;
    lfs_stat(&lfs, "e", &info) => LFS_ERR_NOENT;
    lfs_unmount(&lfs) => 0;
'''
```

</details>