littlefs-project/littlefs #950: renaming symbols

yamt, 2024-03-02:
sometimes a user wants to have multiple copies of littlefs in a binary.

eg.
nuttx embeds littlefs for its filesystem: https://github.com/apache/nuttx/tree/master/fs/littlefs
toywasm also can embed littlefs: https://github.com/yamt/toywasm/blob/master/libwasi_littlefs/README.md
if you enable toywasm on nuttx -> multiple copies of littlefs. (note: nuttx with flat model links everything together)

it would be nice if we can use a different set of symbols to avoid conflicts.
eg. if built with "-DLFS_PREFIX=foo_", provide foo_lfs_mount instead of lfs_mount.
how do you think?

geky, 2024-03-08:
Hi @yamt, thanks for creating an issue.

I think this is better solved with external tooling.

littlefs has the [scripts/changeprefix.py](https://github.com/littlefs-project/littlefs/blob/master/scripts/changeprefix.py), which is how we generated the [v2-prefix](https://github.com/littlefs-project/littlefs/tree/v2-prefix) branch in CI.

Though this is just a glorified sed script:

``` bash
sed 's/\<lfs_/mylfs_/g; s/\<LFS_/MYLFS_/g' lfs.c > lfs.p.c
```

You could even do this at link time with objcopy's --prefix-symbols or --redefine-sym(s).

You could run one of these when you update littlefs, or add it as an additional preprocessing step during compilation.

---

In most cases I would think you would want to actually link to the same littlefs to avoid the ~2x code cost, though I realize this isn't always possible if dependencies need different versions of littlefs.

yamt, 2024-03-08:
> Hi @yamt, thanks for creating an issue.
>
> I think this is better solved with external tooling.
>
> littlefs has the [...]

oh, i wasn't aware of the script.
i will take a look. thank you.

i had concerns about making such mechanical changes as it can make an update difficult.
but as far as it's the "officially supported" way, it's probably fine.

> In most cases I would think you would want to actually link to the same littlefs to avoid the ~2x code cost, though I realize this isn't always possible if dependencies need different versions of littlefs.

in my case, they can have different compile-time options.

geky, 2024-03-08:
> i had concerns about making such mechanical changes as it can make an update difficult.
but as far as it's the "officially supported" way, it's probably fine.

If it's implemented as a preprocessing step, it shouldn't be an update burden in theory.

``` bash
# pretend these are build rules
sed 's/\<lfs_/mylfs_/g; s/\<LFS_/MYLFS_/g' lfs.c > lfs.p.c
cc -c lfs.p.c -o lfs.o
ld lfs.o main.o -o myapp
```

Or

``` bash
# pretend these are build rules
cc -c lfs.c -o lfs.o
objcopy --prefix-symbols=my lfs.o lfs.p.o
ld lfs.p.o main.o -o myapp
```

yamt, 2024-03-11:
ok.

i prefer the sed version because objcopy for mach-o is often broken from my experience.

geky, 2024-03-11:
That's fair. Also more chances for the type-checker to find script issues. It's unfortunate link-time tools are so failure-prone...