littlefs-project/littlefs #952: When are file attributes deleted or not deleted?

RalfEsy, 2024-03-06:
I wonder when file attributes are deleted or not deleted?

So far I've found that file attributes are deleted when the file itself is deleted, but not when the same file is opened for overwriting. It would be nice to get an overview of what operations affect the file attributes (other than setting a file attribute, of course).

Thanks Ralf

geky, 2024-03-08:
Hi @RalfEsy, thanks for creating an issue.

littlefs tries to mimic POSIX here. So attributes remain attached to the metadata entry until the metadata entry ceases to exist.

Note that the definition of LFS_O_CREAT is "Create a file _if it does not exist_". So opening a a file with LFS_O_CREAT does not delete attributes because it doesn't delete the file.

A common pattern in POSIX is to create a file with a different name and use rename to replace the original file. This should clear any attributes on the file in littlefs.

You could also delete and recreate the file, but this would not be atomic.

RalfEsy, 2024-03-08:
Hi @geky not shure if I got that right:

> A common pattern in POSIX is to create a file with a different name and use rename to replace the original file. This should clear any attributes on the file in littlefs.

Do you mean that renaming an existing file will delete all of its attributes? Since I have seen the opposite, the attributes are still available even after a rename?

geky, 2024-03-08:
> Do you mean that renaming an existing file will delete all of its attributes? Since I have seen the opposite, the attributes are still available even after a rename?

Ah, so renaming a file does not delete attributes attached to the file. But if you rename a file on top of another file, it will replace the overlapped file, and in effect replace/delete any attributes that existed on the overlapped file.

For example (pseudocode):

``` c
open("A"); write("A"); close("A");
setattr("A", 'c', 1);
setattr("A", 'd', 2);

open("B"); write("B"); close("B");
setattr("B", 'c', 3);
// no 'd' attr

rename("B", "A");

getattr("A", 'c') => 3;
getattr("A", 'd') => LFS_ERR_NOATTR;
```