littlefs-project/littlefs #962: Expanding superblock deletes File

nobody19, 2024-04-08:
Hi, 
I have some problems with the superblock expansion. When I'm incrementing the bootcounter 500 times, then lfs is reporting "Expanding superblock at rev 1001" and after that the bootcounter file is not anymore available. Currently I'm incrementing the bootcounter in a for loop, to get the error quite fast. 

I'm using lfs v2.9.1 with STM32U5 octaspi flash MX25UM51245G.

[image]
Maybe this is relevated to https://github.com/littlefs-project/littlefs/issues/953

Thanks a lot for any comment which can help me to solve the problem.
nobody

geky, 2024-04-12:
Hi @nobody19, thanks for creating an issue.

I think you're right this is the same issue in #953 (ignoring the #959 red herring).

If you can reproduce this efficiently that's really quite promising.

The next goal is to try to reproduce this locally, to rule out hardware issues and make debugging tractable. So, sorry for the barrage of questions:

1. It looks like this is, or at least started as the example in the README.md.

   Is there many other operations going on? Is the littlefs-specific code sharable? Or could be reduced to sharable code that still reproduces the issue?

2. What is your block_size x block_count?

3. What is your block_cycles?

   Actually, sharing your entire `lfs_config` struct may help reproduce this.

nobody19, 2024-04-15:
Hi @geky,

Thanks a lot for your help. Yes it is more ore less the example from the README. Below I added the lfs structure and in the file should be all function which I'm using to run littleFS. If you have any input, just let me now.

```cpp

const struct lfs_config mylfs_cfg = {
    .context = nullptr,
    // block device operations
    .read  = user_provided_block_device_read,
    .prog  = user_provided_block_device_prog,
    .erase = user_provided_block_device_erase,
    .sync  = user_provided_block_device_sync,

    // block device configuration
    .read_size = 32,
    .prog_size = 256,
    .block_size = 4096,
    .block_count = 16384,
    .block_cycles = 500,
    .cache_size = 4096,
    .lookahead_size = LFS__LOOKAHEAD_CNT,
    .read_buffer = lfs_read_buffer,
    .prog_buffer = lfs_prog_buffer,
    .lookahead_buffer = (uint8_t*)lfs_lookahead_buffer,
    .name_max = LFS_NAME_MAX, 
    .file_max = LFS_FILE_MAX,
    .attr_max = LFS_ATTR_MAX
};
```
[FileSys.txt](https://github.com/littlefs-project/littlefs/files/14974543/FileSys.txt)

geky, 2024-04-16:
Hi @nobody19, thanks for the extra info.

Unfortunately I wasn't able to reproduce this locally. `test_writeBootCountToFile` was able to trigger superblock expansion, but the resulting filesystem continued operating without data loss.

Is it possible that the `BSP_OSPI_NOR_Erase_Block` address should _not_ be multiplied by the block_size? I couldn't really find documentation over this function, though I only searched briefly. Superblock expansion may be the first time when littlefs tries to use blocks > 0 and 1, so I could see it wrapping around and corrupting things.

---

Also just a note:

``` c
file.cfg = &fcfg;
int retOpen =  lfs_file_open(&lfs, &file, "boot_count2", LFS_O_RDWR);
```

The `file.cfg = &fcfg` here doesn't actually do anything. You need to call `lfs_file_opencfg`, any initial state in `lfs_file_t` is ignored:

``` c
int retOpen =  lfs_file_opencfg(&lfs, &file, "boot_count2", LFS_O_RDWR, &fcfg);
```

I don't think this is the source of the problem though, littlefs will fall back to malloc.

nobody19, 2024-04-18:
Hi @geky 

I would also expect some trouble in the Erase function, but it looks that this multiplication by block_size is correct. I checked there the code and also the datasheet. The datasheet is expecting the sector_erase as a 24/32 Bit adress.

[image]

Thanks for the note.

nobody19, 2024-04-18:
I found out why it first "crashes" around 500. It is depending on the block_cycle, which I have set to 500. But the "why" is not clear for me yet.

```cpp
    // Number of erase cycles before littlefs evicts metadata logs and moves
    // the metadata to another block. Suggested values are in the
    // range 100-1000, with large values having better performance at the cost
    // of less consistent wear distribution.
    //
    // Set to -1 to disable block-level wear-leveling.
    int32_t block_cycles;
```

Antecer, 2024-07-09:
I don't know if it is same error, i just press 'res' some times to test it.
It may give this error after a few reboots and then lose the file.

```
	char tmp_str[64];
	lfs_file_t file;
	for(uint8_t i = 0; i < 16; i++)
	{
		memset(tmp_str, 0, 64);
		if(8 != snprintf(tmp_str, 64, "./%02d.txt", i)) continue;
		DebugPrintf("[lfs_open_file] %s\n", tmp_str);
		int status = lfs_file_open(lfs, &file, tmp_str, LFS_O_RDWR | LFS_O_CREAT);
		if(status == 0)
		{
			lfs_file_rewind(lfs, &file);
			DebugPrintf("[lfs_file_write] %s\n", tmp_str);
			lfs_file_write(lfs, &file, tmp_str, strlen(tmp_str));
			lfs_file_close(lfs, &file);
		}
		else
		{
			DebugPrintf("[lfs_open_file] failed: %d\n", status);
		}
	}
	lfs_dir_t ldir;
	if(LFS_ERR_OK == lfs_dir_open(lfs, &ldir, "/"))
	{
		struct lfs_info f_info;
		while(lfs_dir_read(lfs, &ldir, &f_info))
		{
			if(LFS_TYPE_REG == f_info.type)
			{
				DebugPrintf("[lfs_dir_read] size: %ld, name: %s\n", f_info.size, f_info.name);
				lfs_remove(lfs, f_info.name);
			}
		}
		lfs_dir_close(lfs, &ldir);
	}
```

printf log:
```
[lfs_open_file] ./00.txt
[lfs_file_write] ./00.txt
[lfs_open_file] ./01.txt
[lfs_file_write] ./01.txt
[lfs_open_file] ./02.txt
[lfs_file_write] ./02.txt
[lfs_open_file] ./03.txt
[lfs_file_write] ./03.txt
[lfs_open_file] ./04.txt
[lfs_file_write] ./04.txt
[lfs_open_file] ./05.txt
[lfs_file_write] ./05.txt
[lfs_open_file] ./06.txt
[lfs_file_write] ./06.txt
[lfs_open_file] ./07.txt
[lfs_file_write] ./07.txt
[lfs_open_file] ./08.txt
[lfs_file_write] ./08.txt
[lfs_open_file] ./09.txt
[lfs_file_write] ./09.txt
[lfs_open_file] ./10.txt
[lfs_file_write] ./10.txt
[lfs_open_file] ./11.txt
[lfs_file_write] ./11.txt
./libraries/lfs/littlefs/lfs.c:2072:debug: Bad block at 0x0
./libraries/lfs/littlefs/lfs.c:2077:warn: Superblock 0x0 has become unwritable
[lfs_open_file] ./12.txt
[lfs_file_write] ./12.txt
[lfs_open_file] ./13.txt
[lfs_file_write] ./13.txt
[lfs_open_file] ./14.txt
[lfs_file_write] ./14.txt
[lfs_open_file] ./15.txt
[lfs_file_write] ./15.txt
[lfs_dir_read] size: 8, name: 00.txt
[lfs_dir_read] size: 8, name: 01.txt
[lfs_dir_read] size: 8, name: 02.txt
[lfs_dir_read] size: 8, name: 03.txt
[lfs_dir_read] size: 8, name: 04.txt
[lfs_dir_read] size: 8, name: 05.txt
[lfs_dir_read] size: 8, name: 06.txt
[lfs_dir_read] size: 8, name: 12.txt
[lfs_dir_read] size: 8, name: 13.txt
[lfs_dir_read] size: 8, name: 14.txt
```