littlefs-project/littlefs #972: LittleFS Display Directory Content

GitonioDev, 2024-04-23:
Hello all,
I was looking for a simple arduino/c++ solution to use within the NodeMCU module.
I noticed the solution provided was a little complicated for me and maybe related to modifying the LittleFS library itself (which I'm not sure, I'm just guessing).
So I have come up with a simple function for listing all the files inside a directory (recursively). 
I hope it helps future noobies like myself! :D

```
void listAllFilesInDir(String dir_path)
{
	Dir dir = LittleFS.openDir(dir_path);
	while(dir.next()) {
		if (dir.isFile()) {
			// print file names
			Serial.print("File: ");
			Serial.println(dir_path + dir.fileName());
		}
		if (dir.isDirectory()) {
			// print directory names
			Serial.print("Dir: ");
			Serial.println(dir_path + dir.fileName() + "/");
			// recursive file listing inside new directory
			listAllFilesInDir(dir_path + dir.fileName() + "/");
		}
	}
}
```
Use like this:
```
if (LittleFS.begin()) {
	listAllFilesInDir("/");
}
```
The output should be something like this:
```
File: /asset-manifest.json
File: /favicon.ico        
File: /index.html
File: /manifest.json      
File: /robots.txt
Dir: /static/
Dir: /static/css/
File: /static/css/main.1635e87c.chunk.css
Dir: /static/js/
File: /static/js/2.2c405b20.chunk.js
File: /static/js/3.8cb6885c.chunk.js
File: /static/js/main.435818a8.chunk.js
File: /static/js/runtime-main.00e2c71a.js
```

_Originally posted by @jeot in https://github.com/littlefs-project/littlefs/issues/2#issuecomment-881683790_

Hello all, im using the LittleFS libary out of the Arduino ESP32 Core 2.0.15 directly. And right now i am struggling to create a Dir Object cause as it says there should be a Directory Object and then LittleFS.openDir but i just cant do that, it says : "Dir was not declared in this scope" and i dont know how to do it

My Module: ESP32 WROOM

jeot, 2024-04-23:
@GitonioDev 

Hi,

The snippet I originally provided was based on the LittleFS library by Arduino that probably used Littlefs-project (this repo) under the hood (although I'm not sure).

There are lots of examples and use-cases here:

https://github.com/espressif/arduino-esp32/blob/master/libraries/LittleFS/examples/LITTLEFS_test/LITTLEFS_test.ino

GitonioDev, 2024-04-23:
Hi, yeah i seen that method several times but its very slow and not good to display a lot of content the openDir variant looks more efficient and this function is more intended for my case

geky, 2024-04-29:
This may be more useful to open as an issue in https://github.com/espressif/arduino-esp32 or related forums.

It does look like the upstream `listDir` is allocating (three times?) and looking up each path (twice?) during directory traversal, which is a bit wasteful...

But this is the core littlefs repo, we don't really have any control over the framework-specific layers.

closed 2024-04-29 by geky