littlefs-project/littlefs #980: used and free memory

stingdau1206, 2024-05-13:
in file system info, i don't see value that show num of bytes is available, or num of bytes is used
how can i get it ?

geky, 2024-05-13:
Hi @stingdau1206,

[`lfs_fs_size`](https://github.com/littlefs-project/littlefs/blob/d01280e64934a09ba16cac60cf9d3a37e228bb66/lfs.h#L718-L724) returns the number of blocks currently in use. If you multiply this by the block_size the result should be the number of bytes in use.

Because littlefs uses copy-on-write data-structures, the result may sometimes be larger than the actual usage, but it should never be smaller. Hopefully it's still useful for estimating how full the disk is.

stingdau1206, 2024-05-13:
Hi @geky 
thanks for your reply, i'm Looking forward to a more detailed result about disk usage
but, after all, it will be ok for my case

closed 2024-05-13 by stingdau1206