littlefs-project/littlefs #985: How to analye littlefs-binary-file?

Jan9401, 2024-05-23:


Jan9401, 2024-05-23:
I have mouted a partitiont to lillefs, and I have dunped it,I want to analyze what operations have occurred in the file system,Can I use any tools for analysis?

sjzth, 2024-06-01:
Graphviz + CodeViz，You can give it a try

geky, 2024-06-25:
There are a few hacky scripts in [scripts](https://github.com/littlefs-project/littlefs/tree/d01280e64934a09ba16cac60cf9d3a37e228bb66/scripts) that can decode littlefs's metadata:

- scripts/readtree.py - debug a full filesystem tree
- scripts/readmdir.py - debug a single mdir
- scripts/readblock.py - hexdump a block

It would be nice to have more debugging/decoding tools in the future, but this is low priority with everything else going on.

Jan9401, 2024-08-06:
But I couldn't find any documentation on how to use these tools, so I'm not
sure how to use them

geky, 2024-09-21:
Yeah, there is not much documentation at the moment.

`readtree.py` is the easiest, but it needs both the path to an image and the block size:

``` bash
$ ./scripts/readtree.py disk 4096                 
littlefs v2.1                                  data (truncated, if it fits)     
gstate 0x000000000000000000000000                                               
dir "/"                                                                         
mdir {0x1, 0x0} rev 1 (was 0)                                                   
  off       tag       type            id  len  data (truncated)                 
  00000008: 0ff00008  superblock       0    8  6c 69 74 74 6c 65 66 73  littlefs
  00000014: 20100018  inlinestruct     0   24  01 00 02 00 00 10 00 00  ........
```

`readmdir.py` and `readblock.py` are more useful for low-level debugging and need specific block addresses.

There is a bit more info on flags available with `-h`:

``` bash
$ ./scripts/readtree.py -h                        
usage: readtree.py [-h] [-l] [-a] [-T] disk block_size [block1] [block2]        
                                                                                
Dump semantic info about the metadata tree in littlefs                          
                                                                                
positional arguments:                                                           
  disk               File representing the block device.                        
  block_size         Size of a block in bytes.                                  
  block1             Optional first block address for finding the superblock.   
  block2             Optional second block address for finding the superblock.  
                                                                                
options:                                                                        
  -h, --help         show this help message and exit                            
  -l, --log          Show tags in log.                                          
  -a, --all          Show all tags in log, included tags in corrupted commits.  
  -T, --no-truncate  Show the full contents of files/attrs/tags.                
```

But that's all there is right now. These were sort of the bare necessities to get littlefs up and running.

Jan9401, 2024-11-14:
I am very sorry for not replying to your message at first time, your
information is very useful,Thanks for your help.

soliis-jones, 2025-07-31:
Chiming in to see if there are any scripts/external tools that can inspect a disk image binary created by the `v3-alpha` branch? It looks like `readtree.py` does not exist on that branch.