littlefs-project/littlefs #987: Corrupted dir pair at {0x405, 0x406}

p01u9a, 2024-06-01:
I have a small issue with littlefs. Maybe you can point me to the right direction.

I am using STM32F103CBT6 and W25Q128 flash memory.

Flash memory write, read, and erase functions where implemented and are working properly.

In my scenario i am using the default demo that writes a simple counter variable and at  the same time i am writing 100bytes of ascii characters to a second file. After these operations are complete i reboot the mcu and start again (stress testing the process).

After exactly 129 attempts mounting fails and i am presented with the following error.
../littlefs.c:1369:error:Corrupted dir pair at {0x405, 0x406}

Erase function operation:
Function accepts a block number. It will determin the exact location it needs to erase and it will erase it.

Flash write function:
Flash write function will accept the location where the data needs to be written to offsets and so on and it will accept the data that needs to be written. I am taking care of the chunk size it needs to write and i am erasing 4K before writing to the flash within the function.

Flash read function:
Reads the data from the specified location.

At first the issue was related to the configuration lfs. The fail would happen at 5th attempt. And the error that would be displayed was ../littlefs.c:1369:error:Corrupted dir pair at {0x00, 0x01}.

Can you give me an idea how to solve this one ?

p01u9a, 2024-06-02:
[logs-before-crahs.txt](https://github.com/user-attachments/files/15526103/logs-before-crahs.txt)
Logs before formating.

p01u9a, 2024-06-02:
This can be marked as **solved**. It was related to the low level functios that write to the flash memory. What i changed inside of them was before and after setting anything within the w25q128 chip, reading, writing and erasing i am waiting for the busy flag to go to value 0. What happens is sometimes it takes more time than specified by the datasheet to release it.

geky, 2024-06-25:
Hi @p01u9a, interesting issue. Glad you were able to solve it. Timing not matching the datasheet is certainly strange.

closed 2024-06-25 by geky