littlefs-project/littlefs #991: ISR_HARDFAULT() error in Raspberry Pi Pico W !!!

JITU-SYS, 2024-06-07:
Hello , I am using arduino ide 2.0 and my controller is Raspberry Pi Pico W . My code is compiling successfully and it is working just fine for some days . But sometimes while initializing Quicktel EG25 GSM Modem , it gets stuck and after that it is unable to recover from that until I manually reset the board . I have checked and there are no power issues related to GSM module .

Therefore to solve that issue , I used Raspberry Pi debug probe and following are the logs from the debug console .

[image]
[image]
[image]

LittleFS is the file system I am using .

Please guide me immediately to solve this issue .

geky, 2024-06-25:
You will probably get better help if you open this on an actual Pico repo (https://github.com/earlephilhower/arduino-pico, I think?).

This repo is focused on LittleFS's core, which is a step removed from actual hardware.

earlephilhower, 2024-06-25:
I think you can close this, @geky .  The original author posted this here, in the arduino-pico repo, and on the Arduino.cc forums and disappeared.  There was no indication of any issue w/LittleFS or the core, and no MCVE was ever produced.  The pictures of text output from GDB above have nothing point to either, just the usual output when you connect after-the-fact through OpenOCD and GDB.

LittleFS has been absolutely awesome for the RPi Pico and ESP8266 Arduino cores.  Thanks to all the contributors!

geky, 2024-06-25:
This is not the first or last unrelated issue opened here, so no worries. Best of luck in arduino-pico-land.

closed 2024-06-25 by geky