littlefs-project/littlefs #992: Document target hardware class(es) - NOR/NAND/etc

bmcdonnell-fb, 2024-06-10:
Thanks to @geky and all contributors for this project. 😄

I see a couple of issue comments that speak to intended/tested target hardware type, but this does not seem to be explicitly or sufficiently documented.

https://github.com/littlefs-project/littlefs/issues/880#issuecomment-1968175847

> littlefs was written for NOR first and NAND's geometry is quite a bit more challenging.

https://github.com/littlefs-project/littlefs/issues/11#issuecomment-359195144

> LittleFS has only been used on NOR flash, eMMC, SD cards; devices with <4KB erase units, so I'm not sure how it will perform on NAND flash.

Have I missed some documentation that spells this out? I think these items are probably worth at least a mention in the top-level `README.md`.

geky, 2024-06-25:
HI @bmcdonnell-fb,

Ah yes, this could probably be documented better. It would be good to eventually have more thorough evaluation of littlefs on various storage types, but this time could also be spent trying to fix the related performance issues, so prioritization gets a bit muddled.

This has also evolved over time. The comment in #11 notably predates v2, and files can use less than a block if they fit in `cache_size`. You may also not _need_ ECC, but probably want it depending on the device's details.

The original intention of v2 was to work fine on any flash-like technology (or at least any random-access storage), with only different performance characteristics. Unfortunately the lack of thorough testing led to the current state of things on NAND. But with more testing and a better understanding of NAND's issues hopefully things will improve moving forward.

If you want me to document what _future_ technology will be problematic for littlefs, that might be a bit more difficult...

bmcdonnell-fb, 2024-06-25:
> Ah yes, this could probably be documented better. It would be good to eventually have more thorough evaluation of littlefs on various storage types, but this time could also be spent trying to fix the related performance issues, so prioritization gets a bit muddled.

Understandable if other features are taking higher priority over the type of detailed doc that you're envisioning. But I think at least a brief mention in the top level `README.md` of what technologies are known to work, and maybe which are works in progress, would help interested parties to discover littlefs, and to quickly determine whether it may be suitable for their target hw.

>> LittleFS has only been used on NOR flash, eMMC, SD cards; devices with <4KB erase units, so I'm not sure how it will perform on NAND flash.

> The comment in #11 notably predates v2, and files can use less than a block if they fit in `cache_size`.

Good to know. However, in this case, my focus was just to say up front whether littlefs works on NOR flash, eMMC, NAND flash, etc.

It seems most filesystems are not well-suited to NOR flash? And that littlefs was designed with it as a first class citizen?

geky, 2025-11-10:
Just to avoid losing the dialog, copying from https://github.com/littlefs-project/littlefs/issues/1160#issuecomment-3514325782:

> > Has this bit made it into the documentation yet?
>
> @bmcdonnell-fb, no. Feel free to create a PR if you can think of some text that would be valuable.
> 
> I'm hesitant as it's a moving target (see NAND), and I don't want to mislead users into thinking I've actually tested on these devices. Maybe a wiki entry would be more useful? (TODO create a wiki)

geky, 2025-11-10:
> Good to know. However, in this case, my focus was just to say up front whether littlefs works on NOR flash, eMMC, NAND flash, etc.

I think some of the challenge is that "works" is a bit poorly defined.

Technically littlefs should "work" on a tape drive just fine, if you have the patience to wait about a year.

Or, from an actual conversation: "waiting hours for filesystem operations to complete is fine because our device is attached to a pipeline in the middle of nowhere, so what else is it going to do." Embedded engineering is funny in how there's always an exception.

> It seems most filesystems are not well-suited to NOR flash? And that littlefs was designed with it as a first class citizen?

Ah, of the flashes, NOR flash is actually the easier one.

I think ordering by difficulty, you'd have: NVRAM < SD/eMMC < NOR < NAND.

But the tradeoff is roughly density, or cost-per-byte.

(Also, I'm handwaving NVRAM, from what I understand that's actually ~12 different technologies in a trenchcoat. But the differences don't really impact a filesystem.)

Actually, thinking about it now, the differences really just boil down to how big your block size is:

- NVRAM - bs=1 byte (if you squint)
- SD/eMMC - bs=512 bytes
- NOR - bs=~4KiB
- NAND - bs>=~128KiB

The big change in littlefs3 is improving block-size related operations to at most $O(b \log b)$, and this is what makes NAND tractable.

There are other practical limitations that differ between storage types (one-time vs masking progs, cascading progs, erases), but littlefs was already designed for the lowest-common denominator of the storage I know about. I've started referring to this as littlefs's "restricted flash model", and do plan to write a bit more about it.