littlefs-project/littlefs #993: Repeating to create new files when space is out will make littlefs crash

Iammmp93, 2024-06-10:
Hi, folks:
I am using littlefs in nand flash device. When the space is out (actually it's easy to run out of space in nand flash since the erase block is too large like 128k), I try to open new files with different names, like test1, test2, ...test2000, and then littlefs will crash. It looks like there is something wrong in lfs_file_open() which will mess up the metadata.
My test code is very simple.
for (i = 1; i <= 2000; ++i)
{
    memset(file, 0, 32);
    sprintf(file, "%d.txt", i);
    fd = open(file, O_CREAT | O_RDWR);
    if (fd >= 0)
    {
        ret = close(fd);	
    }
}
Does anyone see this issue when use littlefs? Appreciate it if some fix can be provided. Thanks.

geky, 2024-06-25:
Hi @Iammmp93, thanks for creating an issue.

Resuming after an error, usually NOSPC, is unfortunately poorly tested and can't really be relied on. The best option is to unmount+remount the filesystem when this occurs (note `lfs_file_close` still cleans up resources if it errors).

This is something being worked on, but it's not the highest priority at the moment.

Or, well, this is solved/tested in a branch containing a bunch of unrelated work, which makes it difficult to upstream. But with any luck this will land eventually. This is one of those problems that does not have a quick fix.

Iammmp93, 2024-06-25:
Hi, @geky:
Appreciate for the reply. Yes, out of space is an excessive issue with low priority. We are using littlefs on our nand flash, space is a critical one we are facing now, the erase block is too big with 128K bytes. Hope you guys can fix the nand flash space soon. Thanks.