littlefs-project/littlefs #1008: rm file causes program to crash

dyergan, 2024-07-27:
Using EN25QX256A nor flash with system of rt-thread and IC is blueturn(8952F3)

Operation steps:
1.create thirty files,file name (1~30)
2.rm file,one by one.Occasionally the program will crash at this point. Cause lfs_dir_traverse() entering an infinite loop, stack overflow.
3.If stack overflow hanppen, even reboot or burn new program, it will stack overflow.
4.Create a new file, now rm is normal.

Does anyone have a solution?

init :
`struct lfs_config cfg = {
    // block device operations
    .read = awi_lfs_read,
    .prog = awi_lfs_prog,
    .erase = awi_lfs_erase,
    .sync = awi_lfs_sync,

    // block device configuration
    .read_size = READ_SIZE,                      //16
    .prog_size = PROG_SIZE,                    //16
    .block_size = BLOCK_SIZE,                 //16
    .block_count = BLOCK_COUNT,        //4096
    .cache_size = CACHE_SIZE,                 //128
    .lookahead_size = LOOKAHEAD_SIZE,  //16
    .block_cycles = 500,

    .lookahead_buffer = lookahead_buffer,
    .read_buffer = read_buffer,
    .prog_buffer = prog_buffer,
};
`

dyergan, 2024-07-29:
add some printf in lfs_dir_traverse:

```
static int lfs_dir_traverse(lfs_t *lfs,
        const lfs_mdir_t *dir, lfs_off_t off, lfs_tag_t ptag,
        const struct lfs_mattr *attrs, int attrcount,
        lfs_tag_t tmask, lfs_tag_t ttag,
        uint16_t begin, uint16_t end, int16_t diff,
        int (*cb)(void *data, lfs_tag_t tag, const void *buffer), void *data) {
    // This function in inherently recursive, but bounded. To allow tool-based
    // analysis without unnecessary code-cost we use an explicit stack
    WDT_CLR();
    struct lfs_dir_traverse stack[LFS_DIR_TRAVERSE_DEPTH-1];
    unsigned sp = 0;
    int res;

    // iterate over directory and attrs
    lfs_tag_t tag;
    const void *buffer;
    struct lfs_diskoff disk = {0};
    dir_times++;
    printf("line:%d times:%d\n", __LINE__, dir_times);
    while (true) {
		WDT_CLR();
        {
            printf("line:%d off+lfs_tag_dsize(ptag):%d dir->off: %d\n", __LINE__, off+lfs_tag_dsize(ptag), dir->off);
            if (off+lfs_tag_dsize(ptag) < dir->off) {
                off += lfs_tag_dsize(ptag);
                int err = lfs_bd_read(lfs,
                        NULL, &lfs->rcache, sizeof(tag),
                        dir->pair[0], off, &tag, sizeof(tag));
                if (err) {
                    return err;
                }

                tag = (lfs_frombe32(tag) ^ ptag) | 0x80000000;
                disk.block = dir->pair[0];
                disk.off = off+sizeof(lfs_tag_t);
                buffer = &disk;
                ptag = tag;
                printf("line:%d tag:%d  disk.block :%d dir->off: %d buffer :%d\n", __LINE__, tag, disk.block, dir->off, buffer);
            } else if (attrcount > 0) {
                tag = attrs[0].tag;
                buffer = attrs[0].buffer;
                attrs += 1;
                attrcount -= 1;
            } else {
                // finished traversal, pop from stack?
                res = 0;
                break;
            }

            // do we need to filter?
            lfs_tag_t mask = LFS_MKTAG(0x7ff, 0, 0);
            if ((mask & tmask & tag) != (mask & tmask & ttag)) {
                continue;
            }
            printf("line:%d lfs_tag_id(tmask):%d sp: %d\n", __LINE__, lfs_tag_id(tmask), sp);
            if (lfs_tag_id(tmask) != 0) {
                LFS_ASSERT(sp < LFS_DIR_TRAVERSE_DEPTH);
                // recurse, scan for duplicates, and update tag based on
                // creates/deletes
                stack[sp] = (struct lfs_dir_traverse){
                    .dir        = dir,
                    .off        = off,
                    .ptag       = ptag,
                    .attrs      = attrs,
                    .attrcount  = attrcount,
                    .tmask      = tmask,
                    .ttag       = ttag,
[... 289 lines cut ...]
line:935 off+lfs_tag_dsize(ptag):269 dir->off: 4096
line:950 tag:-2146429950  disk.block :1 dir->off: 4096 buffer :78772
line:967 lfs_tag_id(tmask):0 sp: 1
line:1004 lfs_tag_id(tmask):0
line:1009 lfs_tag_type3(tag)1 sp: 1
line:935 off+lfs_tag_dsize(ptag):275 dir->off: 4096
line:950 tag:-1608510456  disk.block :1 dir->off: 4096 buffer :78772
line:967 lfs_tag_id(tmask):0 sp: 1
line:1004 lfs_tag_id(tmask):0
line:1009 lfs_tag_type3(tag)514 sp: 1
line:935 off+lfs_tag_dsize(ptag):287 dir->off: 4096
line:950 tag:-2146428926  disk.block :1 dir->off: 4096 buffer :78772
line:967 lfs_tag_id(tmask):0 sp: 1
line:1004 lfs_tag_id(tmask):0
line:1009 lfs_tag_type3(tag)1 sp: 1
line:935 off+lfs_tag_dsize(ptag):293 dir->off: 4096
line:950 tag:-1608509432  disk.block :1 dir->off: 4096 buffer :78772
line:967 lfs_tag_id(tmask):0 sp: 1
line:1004 lfs_tag_id(tmask):0
line:1009 lfs_tag_type3(tag)514 sp: 1
line:935 off+lfs_tag_dsize(ptag):305 dir->off: 4096
line:950 tag:-536871928  disk.block :1 dir->off: 4096 buffer :78772
line:967 lfs_tag_id(tmask):0 sp: 1
line:1004 lfs_tag_id(tmask):0
line:1009 lfs_tag_type3(tag)1535 sp: 1 ............

dyergan, 2024-07-29:
[image]
In picture , when rm two file, step of 1 ptag  > dir->off 4096,now rm is normal.But   step 2  ptag < dir->off, and dir->off is 4096 ,now rm entering an infinite loop.

geky, 2024-09-27:
Hi @dyergan, thanks for creating an issue, sorry for the late response.

If you're experiencing a stack overflow, is it possible there is just not enough stack for littlefs to operate?

littlefs uses a bounded RAM, but it can still be relatively large. Currently CI says it needs ~1.5 KiB of stack.

> In picture , when rm two file, step of 1 ptag > dir->off 4096,now rm is normal.But step 2 ptag < dir->off, and dir->off is 4096 ,now rm entering an infinite loop.

If there was a stack overflow, the in-RAM filesystem state could be corrupted. Or, it's possible that it's not an infinite loop, but just a very long running loop due to performance issues with littlefs and metadata compaction.