littlefs-project/littlefs #1011: Getting LFS_ERR_NOSPC way before memory is full

pedromarinho94, 2024-07-29:
I am having some issues with my littlefs (v2.9.3) implementation using a W25Q128 via QSPI.

I am getting the following error way before the flash memory is full. (approx 98% free space left to write)

**1722291908 [ERROR] lfs.c:689: ./fw-fdk/3rd-party/littlefs/lfs.c:689:error: No more free space 0x3**

I have exhausted all possibilities. I have written a similar application with littlefs-python that replicates what I'm doing and it writes 98%99% of all the possible bytes.

What could it be?

geky, 2024-09-27:
Hi @pedromarinho94, thanks for creating an issue, sorry about the late response.

That is a strange, the presence of the error log suggests littlefs truly can't find a free block.

Are there any other logging statements with LFS_DEBUG/WARN? do you have asserts enabled? It's possible writes are failing and after trying every block littlefs gives up.

What is your configuration?

pedromarinho94, 2024-09-27:
@geky I'm sorry. We had a bug in our SPI driver in which we were always deleting block 0. It's fixed now.

geky, 2024-09-27:
Ah, glad to hear you found the issue.

closed 2024-09-27 by geky