littlefs-project/littlefs #1012: why is it erasing blocks on every string write ?

vinodsdewa, 2024-07-30:
`
#include "lfs.h"

lfs_t lfs;
lfs_file_t file;

#define BLOCK_COUNT 10
#define BLOCK_SIZE (1024*8)

uint8_t buff[BLOCK_SIZE * BLOCK_COUNT];

static int lfs_hal_read(const struct lfs_config *c, lfs_block_t block, lfs_off_t off, void *buffer, lfs_size_t size)
{
    memcpy(buffer, buff + (BLOCK_SIZE*block) + off, size);
    return 0;
}

static int  lfs_hal_prog(const struct lfs_config *c, lfs_block_t block, lfs_off_t off, const void *buffer, lfs_size_t size)
{
    memcpy(buff + (BLOCK_SIZE*block) + off, buffer, size);
    return 0;
}

static int  lfs_hal_erase(const struct lfs_config *c, lfs_block_t block)
{
    memset( buff + (BLOCK_SIZE * block), 0xff, BLOCK_SIZE);
    printf("erasing block %d\n", block);
    return 0;
}

static int lfs_hal_sync(const struct lfs_config *c)
{
    printf("sync\n");
    return 0;
}

// configuration of the filesystem is provided by this struct
const struct lfs_config cfg = {
    // block device operations
    .read  = lfs_hal_read,
    .prog  = lfs_hal_prog,
    .erase = lfs_hal_erase,
    .sync  = lfs_hal_sync,

    // block device configuration
    .read_size = 16,
    .prog_size = 16,
    .block_size = BLOCK_SIZE,
    .block_count = BLOCK_COUNT,
    .cache_size = 16,
    .lookahead_size = 16,
    .block_cycles = 500,
};

void main(void)
{
	int err = lfs_mount(&lfs, &cfg);

    // reformat if we can't mount the filesystem
    // this should only happen on the first boot
    if (err) {
        lfs_format(&lfs, &cfg);
        lfs_mount(&lfs, &cfg);
    }

    uint32_t boot_count = 0;
    int val  = 0;
    char url_read[100];
    char url_write[100];

    for(int i = 0; i < 10; i++) {

        memset(url_read, 0, sizeof(url_read));

        lfs_file_open(&lfs, &file, "MQTT_URL", LFS_O_RDWR | LFS_O_CREAT);
        lfs_file_read(&lfs, &file, url_read, 99);
        printf("read data = %s\n", url_read);

        sprintf(url_write, "http://broker.hivemq.com/%d", i);

        lfs_file_rewind(&lfs, &file);
        lfs_file_write(&lfs, &file, url_write, strlen(url_write));
        lfs_file_close(&lfs, &file);

    }
}
`

`F:\PROJECT\scratch\littlefs\lfs.c:1370:error: Corrupted dir pair at {0x0, 0x1}
erasing block 0
sync
erasing block 1
sync
sync
read data =
erasing block 8
sync
sync
read data = http://broker.hivemq.com/0
erasing block 9
sync
sync
read data = http://broker.hivemq.com/1
erasing block 2
sync
sync
read data = http://broker.hivemq.com/2
erasing block 3
sync
sync
read data = http://broker.hivemq.com/3
erasing block 4
sync
sync
read data = http://broker.hivemq.com/4
erasing block 5
sync
sync
read data = http://broker.hivemq.com/5
erasing block 6
sync
sync
read data = http://broker.hivemq.com/6
erasing block 7
sync
sync
read data = http://broker.hivemq.com/7
erasing block 8
sync
sync
read data = http://broker.hivemq.com/8
erasing block 9
sync
sync

Process returned 0 (0x0)   execution time : 0.009 s
Press any key to continue.`

Trying to test it on RAM buffer, but even lot of memory is there, it is still erasing each block after writing just a small string!! 
If I increase all prog size, read size, cache size etc to 128 or so, it works, but this should not be the expected behaviour and this can damage my flash memory due to this buggy erase behaviour !! Any clue ?

pedromarinho94, 2024-08-01:
I think this is because of the APPEND flag

vinodsdewa, 2024-08-02:
> I think this is because of the APPEND flag

Which is the append flag?

bauerjj, 2024-08-29:
> Trying to test it on RAM buffer, but even lot of memory is there, it is still erasing each block after writing just a small string!!
If I increase all prog size, read size, cache size etc to 128 or so, it works, but this should not be the expected behaviour and this can damage my flash memory due to this buggy erase behaviour !! Any clue ?

I have the same "issue", however this is actually just how `littlefs` works. See https://github.com/littlefs-project/littlefs/issues/905#issuecomment-1862219166 

> I think this is the file padding issue, seen also in https://github.com/littlefs-project/littlefs/issues/862. Long story short littlefs currently does not track erased state/alignment in files, so appending to files always requires rewriting (with erase) the last block in the file. Since your blocks contain the whole file, this means littlefs is rewriting the whole file.

> You shouldn't see this behavior if you don't call lfs_file_sync. But I realize that is a non-option for a number of use cases.

> The real fix (tracking erased-state) is in the works, but due to other things it is getting rolled up with bigger changes to the file data structure. Which unfortunately means it will take some time before it's usable... On the plus side the erased-state checksums necessary to make this work are passing testing on this experimental branch... Though I'm not sure that's consolation...

and also https://github.com/littlefs-project/littlefs/issues/862 

Let me know if there is ever a solution where `littefs` does NOT erase a block after any `append` operation