littlefs-project/littlefs #1014: How to implement these functions at the bottom of file IO using littlefs

shadowYYMM, 2024-08-06:
I use RT-THREAD nano version system, after porting littlefs to nor flash, I can not realize the open, close, read, write functions in file IO

How to implement these functions at the bottom of file IO using littlefs：
int _open_r(struct _reent *ptr, const char *file, int flags, int mode)；
int _close_r(struct _reent *ptr, int fd)；
_ssize_t _read_r(struct _reent *ptr, int fd, void *buf, size_t nbytes)；
_ssize_t _write_r(struct _reent *ptr, int fd, const void *buf, size_t nbytes)；

wdfk-prog, 2024-08-06:
> I use RT-THREAD nano version system, after porting littlefs to nor flash, I can not realize the open, close, read, write functions in file IO我用的是RT-THREAD nano版本的系统，把littlefs移植到nor flash后，无法实现文件IO的开、关、读、写功能
> 
> How to implement these functions at the bottom of file IO using littlefs：如何使用 littlefs 在文件 IO 的底部实现这些函数： `int _open_r(struct _reent *ptr, const char *file, int flags, int mode)；` `int _close_r(struct _reent *ptr, int fd)；` `_ssize_t _read_r(struct _reent *ptr, int fd, void *buf, size_t nbytes)；` `_ssize_t _write_r(struct _reent *ptr, int fd, const void *buf, size_t nbytes)；`

- Reference package Implementation,
>https://github.com/RT-Thread-packages/littlefs/blob/master/dfs_lfs.c

shadowYYMM, 2024-08-06:
> > 我用的是RT-THREAD nano版本的系统，把littlefs移植到nor flash后，我无法实现IO文件中的打开、关闭、读、写功能我用的是RT-THREAD nano版本的系统，把littlefs移植到nor flash后，无法实现文件IO的开、关、读、写功能
> > 如何使用 littlefs 在文件 IO 的底部实现这些函数：如何使用 littlefs 在文件 IO 的底部实现这些函数：`int _open_r(struct _reent *ptr, const char *file, int flags, int mode)；` `int _close_r(struct _reent *ptr, int fd)；` `_ssize_t _read_r(struct _reent *ptr, int fd, void *buf, size_t nbytes)；` `_ssize_t _write_r(struct _reent *ptr, int fd, const void *buf, size_t nbytes)；`
> 
> * 参考包实现，
> 
> > https://github.com/RT-Thread-packages/littlefs/blob/master/dfs_lfs.c

The nano version does not have a dfs component

`/usr/lib/gcc/arm-none-eabi/13.2.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/13.2.0/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libg.a(libc_a-closer.o): in function `_close_r':
/build/arm-none-eabi-newlib/src/build-newlib/arm-none-eabi/thumb/v7e-m+fp/hard/newlib/../../../../../../newlib-4.3.0.20230120/newlib/libc/reent/closer.c:47:(.text._close_r+0xc): undefined reference to `_close'
/usr/lib/gcc/arm-none-eabi/13.2.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/13.2.0/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libg.a(libc_a-lseekr.o): in function `_lseek_r':
/build/arm-none-eabi-newlib/src/build-newlib/arm-none-eabi/thumb/v7e-m+fp/hard/newlib/../../../../../../newlib-4.3.0.20230120/newlib/libc/reent/lseekr.c:49:(.text._lseek_r+0x14): undefined reference to `_lseek'
/usr/lib/gcc/arm-none-eabi/13.2.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/13.2.0/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libg.a(libc_a-readr.o): in function `_read_r':
/build/arm-none-eabi-newlib/src/build-newlib/arm-none-eabi/thumb/v7e-m+fp/hard/newlib/../../../../../../newlib-4.3.0.20230120/newlib/libc/reent/readr.c:49:(.text._read_r+0x14): undefined reference to `_read'
/usr/lib/gcc/arm-none-eabi/13.2.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/13.2.0/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libg.a(libc_a-writer.o): in function `_write_r':
/build/arm-none-eabi-newlib/src/build-newlib/arm-none-eabi/thumb/v7e-m+fp/hard/newlib/../../../../../../newlib-4.3.0.20230120/newlib/libc/reent/writer.c:49:(.text._write_r+0x14): undefined reference to `_write'
/usr/lib/gcc/arm-none-eabi/13.2.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/13.2.0/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libg.a(libc_a-fstatr.o): in function `_fstat_r':
/build/arm-none-eabi-newlib/src/build-newlib/arm-none-eabi/thumb/v7e-m+fp/hard/newlib/../../../../../../newlib-4.3.0.20230120/newlib/libc/reent/fstatr.c:55:(.text._fstat_r+0x12): undefined reference to `_fstat'
/usr/lib/gcc/arm-none-eabi/13.2.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/13.2.0/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libg.a(libc_a-isattyr.o): in function `_isatty_r':
/build/arm-none-eabi-newlib/src/build-newlib/arm-none-eabi/thumb/v7e-m+fp/hard/newlib/../../../../../../newlib-4.3.0.20230120/newlib/libc/reent/isattyr.c:52:(.text._isatty_r+0xc): undefined reference to `_isatty'
/usr/lib/gcc/arm-none-eabi/13.2.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/13.2.0/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libg.a(libc_a-signalr.o): in function `_kill_r':
/build/arm-none-eabi-newlib/src/build-newlib/arm-none-eabi/thumb/v7e-m+fp/hard/newlib/../../../../../../newlib-4.3.0.20230120/newlib/libc/reent/signalr.c:53:(.text._kill_r+0x12): undefined reference to `_kill'
/usr/lib/gcc/arm-none-eabi/13.2.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/13.2.0/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libg.a(libc_a-signalr.o): in function `_getpid_r':
/build/arm-none-eabi-newlib/src/build-newlib/arm-none-eabi/thumb/v7e-m+fp/hard/newlib/../../../../../../newlib-4.3.0.20230120/newlib/libc/reent/signalr.c:83:(.text._getpid_r+0x0): undefined reference to `_getpid'`

wdfk-prog, 2024-08-07:
> The nano version does not have a dfs component
>
> `/usr/lib/gcc/arm-none-eabi/13.2.0/../../../../arm-none-eabi/bin/ld: [...]

Reference only,nano can only write its own;

shadowYYMM, 2024-08-07:
> Reference only,nano can only write its own;

Thanks for your answer.I'm trying to write it