littlefs-project/littlefs #1016: STM32H7 Internal Flash No more free space

FarukSozuer, 2024-08-14:
Hello everyone,
I'm working on littleFS integration using STM32H7's Internal Flash. I have no problems creating, deleting files, etc. but I encounter a problem when I want to create a log file. If I were to explain all the details in order;

1. My littleFS configs are as follows. The internal flash of the processor I use has a total of 16 sectors and each sector is 128KB. I use the last 2 sectors for littleFS. Accordingly, my config file is as follows.
`lfsConfig.read_size = 256;
 lfsConfig.prog_size = 256;
lfsConfig.block_size = 0x00020000; // 128kB
lfsConfig.block_count = 2;
lfsConfig.cache_size = 1024;
lfsConfig.lookhead_size = 8;
lfsConfig.block_cycles =1000;

lfsConfig.read = internalFlashRead;
lfsConfigprog = internalFlashProgram;
lfsConfig.erase = internalFlashErase;
lfsConfig.sync = flash_sync;`

2. Then I create a new file and write to this file at certain intervals. The function I use in this is below.
`void initializeStorage(void)
{
       int32_t error;

      error = lfs_mount(&lfs,&lfsConfig);
      if(error != LFS_ERR_OK)
      {
       error = lfs_format(&lfs,&lfsConfig);
       error = lfs_mount(&lfs,lfsConfig);
      
       error = lfs_file_open(&lfs,&file, "counter.txt", LFS_O_RDWR | LFS_O_CREAT | LFS_O_APPEND);
       error = lfs_close(&lfs,&file);
      }
}`
`int32_t updateStorage(void)
{
      int32_t error;
      char log_file[10];
      
      log_file[0] = 'A';
      log_file[1] = 'A';
      log_file[2] = 'A';
      log_file[3] = 'A';
      log_file[4] = 'A';
      log_file[5] = 'A';
      log_file[6] = 'A';
      log_file[7] = 'A';
      log_file[8] = 'A';
      log_file[9] = '\r';
      log_file[10] = '\n';
      
      error = lfs_file_open(&lfs,&file, "counter.txt", LFS_O_RDWR );
      seek_size = lfs_file_size(&lfs,&file);
      error = lfs_file_seek(&lfs,&file,seek_size,LFS_SEEK_SET);
      error = lfs_file_write(&lfs,&file&log_file,sizeof(log_file));
      error = lfs_file_close(&lfs,&file);

      return error;
}`
3. The example I gave above works fine. But when the file size reaches 1024 bytes, I start getting the No More Free space error. I think this is directly related to cache_size. But I'm not sure how to overcome it.

pseudotronics, 2024-08-14:
The nature of block based file systems means that what you are trying to do is impossible.  Two blocks is far too few to store the relevant metadata needed yet alone actual files.

The block size ST provides is really intended to store flash images ie put a bootloader in the first sector and then store application firmware that is field upgradeable in the other sectors.

FarukSozuer, 2024-08-14:
I am aware of this. I wanted to ask again to confirm. I think what I need is a memory unit divided into 1K sectors. for example winbond qspi flash.

geky, 2024-09-27:
@pseudotronics is mostly right, but littlefs does support inline files in metadata blocks which changes things. One design goal is for 2-block filesystems to be possible.

The main catch right now is littlefs needs to be able to back inline files in RAM, which is indirectly controlled by `cache_size`. This is probably why you were able to write smaller files, but ran into problems with the log file.

At 128 KiB sectors the RAM requirement may be problematic. There is work ongoing to remove the RAM requirement, but it is still an incomplete feature.

FarukSozuer, 2024-09-29:
@geky Thanks for reply. I will follow the developments. For now, I am progressing the issue with an external flash.