littlefs-project/littlefs #1024: Parsing order in a folder

tototrix23, 2024-09-16:
Hi everyone,

It's not an issue but a question. When you create hundred of files in a folder and you want to parse it: 
what is the parsing order ? The oldest created file to the newest ? Is it true all the time ?

Best regards.

geky, 2024-09-17:
Hi @tototrix23, thanks for creating an issue.

The order `lfs_dir_read` returns files is a bit weird. It's almost alphabetical, but with larger file names being returned first.

Originally the ordering was intended to be strictly alphabetical, but a bug in the comparison logic went unnoticed. Backwards compatible with existing disks makes this difficult to change without a minor/major on-disk version bump. More info here: https://github.com/littlefs-project/littlefs/issues/923.

Still, this ordering can be useful if you want to intentionally order files with same-sized filenames.

cryptoAlgorithm, 2025-11-11:
Hi @geky 

Could I clarify if the larger file names being returned first only applies to names which would otherwise be equal, save for their length? If my understanding of the bug is accurate, sorting order of names of different lengths is correct except when they only differ in length.

i.e. 
`1234` < `234` (correct sorting)
`11111` > `111` (incorrect, longer string first)

Is my understanding correct?