littlefs-project/littlefs #1028: Case sensitivity

UKTailwind, 2024-10-01:
Is there anyway to turn off case sensitivity for filenames? I'm running code with LittleFS on the flash "disk" and FatFS on the SDcard. FatFS is not case sensitive and the differencemakes it confusing for users

geky, 2024-10-02:
Hi @UKTailwind, thanks for creating an issue.

The problem is you don't really "turn off" case sensitivity. Case sensitivity is just comparing bytes, while case _insensitivity_ is a bit more complicated ([_much_ more complicated](https://www.b-list.org/weblog/2018/nov/26/case/) if you want to support unicode).

I realize this is opinion territory, but I think we really need to move away from case insensitivity in filesystems. Case sensitivity is the norm in POSIX/Linux, and I've used Windows enough to know case insensitivity can cause some really weird problems for applications.

That being said, one way you could emulate case insensitivity is by mapping everything to lower case (or upper case), and store the "display name" as a [custom attribute](https://github.com/littlefs-project/littlefs/blob/d01280e64934a09ba16cac60cf9d3a37e228bb66/lfs.h#L544-L552). I think this would cover the use case.

geky, 2024-10-02:
Alternatively, consider only providing your application for languages without letter casing, such as Chinese or Korean. This avoids the problem entirely. (joking)