littlefs-project/littlefs #1043: Create files using the same method, some files are 0x201 (inline) type and cannot be written

Ldd309, 2024-11-15:
Hello,
In the current file system, the same method is used to create files, but several of the files cannot be written, and the size of the file is also 0, and the data in the file system is parsed, and the file type that cannot be written is 0x201 (inline), and the other file types that can be written normally are 0x202 (CTZ-struct).

Littlefs version is used: V2.5.0

How to reproduce: Unfortunately, there is no way to reproduce this problem reliably, and only one device in the same code environment has this problem.

Please ask what might be causing this issue, and is there anything that can be done to fix or avoid it, or in which version it has been fixed.

[image]
[image]

Thank you for your time and help!

Ldd309, 2024-11-22:
Hello @geky,
Thanks for your reply, but the device did not lose power during the file writing.

geky, 2024-11-23:
That is strange then. Zero-length files could also be created if a file was opened but never closed. Or if there was memory corruption.

But if these files can't be written, that is curious. Do you can an error when trying to write to the files?

The fact that these files are all next to each others suggests they could be in the same metadata-log, which might mean the metadata-log was corrupted somehow.

Ldd309, 2024-11-25:
I tried dump the file system data on the device, and mounted it on linux through lfs-fuse for test, linux can write these data normally. I think there may be a problem with flash?

geky, 2024-11-25:
That would be explained by a block returning incorrect data. Was this device under heavy use? If not maybe it was a manufacturing defect?

littlefs does extend the lifetime of the device by distributing wear it across all blocks in the device, but littlefs doesn't have mechanisms to detect or correct errors when they do happen. There's some long term plans to add error detection+correction, but they are currently just plans.

Some users have had success putting checksums in the block device layer to detect errors, if you need to protect against this type of failure: https://github.com/littlefs-project/littlefs/issues/843