littlefs-project/littlefs #1054: Poor append-only performance with large files.

andyp1per, 2024-12-16:
My setup:

W25N01GV, 2Gbit chip, with 2k pages and 128k blocks
```
    fs_cfg.read_size = page_size;
    fs_cfg.prog_size = page_size;
    fs_cfg.block_size = block_size;
    fs_cfg.block_count = block_count;
    fs_cfg.metadata_max = page_size;
    fs_cfg.lookahead_size = 128;
    fs_cfg.cache_size = page_size;
```
I am trying to write an append-only log file at about 318kB/s which is 158 pages/s
I am only syncing the file *per-block* by using the algorithm in https://github.com/littlefs-project/littlefs/issues/564#issuecomment-2363032827 

My logging subsystem is doing *no* reads (I instrumented read() to check) but I see pages being read at about 274/s.
Worse the write speed slows down as the file gets bigger - going down to 90kB/s. Start a new file and the write speed bounces back up.
sync takes about 11ms, which is slow but not awful.

So my questions:
- what in littlefs is reading all the pages? It's going to severely limit the amount of writing I can do
- will making the cache bigger help? If so at what cost?
- Why the slowdown over time related to filesize.

andyp1per, 2024-12-18:
I added this which helps: https://github.com/littlefs-project/littlefs/pull/1056
The other reason is that checking the fs size is very expensive so I have stopped us doing that.

andyp1per, 2024-12-19:
> * Increasing the size of the lookahead buffer (`lookahead_size`) will make gc run less often, though it's not possible to completely avoid gc.
> 
> * Increasing the `block_size` means fewer blocks for the same amount of storage, which can soften the gc bottleneck.

Thanks for the reply:
- What is a good size for lookahead_size? I am currently at 128.
- The block size can be bigger than the physical erase size? What are the downsides of setting it bigger? If its bigger how does littlefs know what the erase size is?

geky, 2024-12-19:
> What is a good size for lookahead_size? I am currently at 128.

I don't think there's an easy answer without benchmarking on the device. It's a tradeoff of RAM to frequency of garbage collection, though there is no benefit to a lookahead larger than block_count/8.

> The block size can be bigger than the physical erase size? What are the downsides of setting it bigger?

The downsides of bigger blocks are 1. less block granularity, so things like small non-inlined or unaligned files can end up wasting more space, and 2. more expensive in-block operations, specifically metadata logs.

This is another littlefs performance bottleneck in that metadata compaction also grows $O(n^2)$: https://github.com/littlefs-project/littlefs/issues/214. At least here, `metadata_max` offers a workaround.

> If its bigger how does littlefs know what the erase size is?

The neat part is littlefs doesn't really care about the physical erase size. It's up to the `read/prog/erase` callbacks to map correctly from logical block to physical block.

Though at some point I think it would be a good idea to add `erase_size` to the config. If anything just to assert on alignment issues and make it more clear to users that the logical block size is not tied to the flash. But this is low priority since it would probably mean API breakage.