littlefs-project/littlefs #1063: lfs_alloc() blocks on full filesystem.

ddomnik, 2025-01-11:
This issue is derived from this Espressif port: https://github.com/joltwallet/esp_littlefs/issues/213 But I think it most likely fits better here.

Usually on a full filesystem littleFS returns from fwrite() within a few ms and prints:
E (2818207) esp_littlefs: ./managed_components/joltwallet__littlefs/src/littlefs/lfs.c:689:error: No more free space 0x48d

However, it could happen that the library gets "stuck" for a very long time (multiple seconds or even minutes - prob. depending on partition size). This happens exactly in this while loop. (The second while in lfs_alloc())
https://github.com/littlefs-project/littlefs/blob/d01280e64934a09ba16cac60cf9d3a37e228bb66/lfs.c#L656

The problem is, that usually every program has watchdogs. Espressifs default watchdog has a maximum of 60 seconds, which seems to be insufficient in rare cases. [maximum wdt timeout](https://docs.espressif.com/projects/esp-idf/en/v5.1.4/esp32/api-reference/kconfig.html#config-esp-task-wdt-timeout-s)

Maybe some experts know how to handle this issue.

Based on my current understanding of the issue, a very simple thought would be to give littleFs it's own watchdog that can be set. This allows to "catch" the issue on application level rather than system level.

BrianPugh, 2025-02-12:
> One option is to reset the watchdog in the low-level bd read/prog/erase functions.

That's an idea. We could add an additional layer of callbacks in `esp_littlefs` by adding 4x function handles to `esp_vfs_littlefs_conf_t` corresponding to read/write/erase/sync. We could have them be conditional based off macros so that the structure is only larger for users that want to use the callbacks.

Additionally, we could just make a configuration that enables/disables automatic calling of `esp_task_wdt_reset` in the read/write/erase/sync functions.

If either of these is something we want to pursue, please follow up in https://github.com/joltwallet/esp_littlefs/issues/213

geky, 2025-02-13:
It probably won't help you now, but eventually I'd like to redesign the bd API to take ctx directly. It's an unfortunate missed opportunity that the current API makes composability difficult...

But the bd redesign is currently mid-priority in a sea of high-priority todos...