littlefs-project/littlefs #1066: File too large when writing

BlazejKrysztofiak, 2025-01-29:
Hi

I am using littleFs with W25Q64JV NAND memory and I am trying to save about 1.7M byte data to file by appending it with smaller data chunks (about 20 bytes each). On some point lfs_file_write() function is returning error code LFS_ERR_FBIG.

My configuration:

#define EXTERNAL_FLASH_PAGE_SIZE        ( 256 )
#define EXTERNAL_FLASH_SECTOR_SIZE      ( 4096 )
#define EXTERNAL_FLASH_SECTORS_IN_BLOCK ( 16 )
#define EXTERNAL_FLASH_BLOCKS_NUMBER    ( 128 )

`static const struct lfs_config lfsCfg =
{
    .read             = Read,
    .prog             = Prog,
    .erase            = Erase,
    .sync             = Sync,
    .read_size        = 1,
    .prog_size        = 1,
    .block_size       = EXTERNAL_FLASH_SECTOR_SIZE,
    .block_count      = EXTERNAL_FLASH_SECTORS_IN_BLOCK * EXTERNAL_FLASH_BLOCKS_NUMBER,
    .block_cycles     = 100,
    .cache_size       = EXTERNAL_FLASH_SECTOR_SIZE,
    .lookahead_size   = EXTERNAL_FLASH_SECTORS_IN_BLOCK * EXTERNAL_FLASH_BLOCKS_NUMBER / 8,
    .compact_thresh   = 0, // Fefault value
    .read_buffer      = ( void* )readBuffer,
    .prog_buffer      = ( void* )progBuffer,
    .lookahead_buffer = ( void* )lookaheadBuffer,
    .name_max         = FILE_SYSTEM_NAME_LENGTH_MAX,
    .file_max         = 0, // Default value
    .attr_max         = 0, // Default value
    .metadata_max     = 0, // Default value
    .inline_max       = 0, // Default value
};`

My sync function is not implemented, it basically returns LFS_ERR_OK.

Do I try to write too much data into single file? Maybe I should close the file somewhere between writes and then reopen it?

geky, 2025-02-03:
Hi @BlazejKrysztofiak, thanks for creating an issue.

That is curious. I've not seen any issues related to LFS_ERR_FBIG before, the relevant logic is honestly pretty simple:

https://github.com/littlefs-project/littlefs/blob/0494ce7169f06a734a7bd7585f49a9fa91fa7318/lfs.c#L3673-L3676

Is it possible `file->pos` is getting corrupt somehow? Are you seeking around in the file?

I would also check that the file is open and hasn't been closed before the write call. Enabling asserts would help catch this, if they're disabled.

geky, 2025-02-03:
It's also worth mentioning running out of space is reported as LFS_ERR_NOSPC, which is what makes LFS_ERR_FBIG curious. It's only possible to get this if you seek close to the file limit, which is INT_MAX by default:

https://github.com/littlefs-project/littlefs/blob/0494ce7169f06a734a7bd7585f49a9fa91fa7318/lfs.h#L54-L59

Unless you are on a 16-bit system?

BlazejKrysztofiak, 2025-02-04:
Hi @geky 

Thank you for answering.

I have asserts enabled. I am not seeking around the file as I have it opened in "append" mode and I am only adding new data to the end of the file. I am using 32 bit STM32G474.

I did not checked size of the file after I got error because my code is removing the file if all data was not saved correctly. Probably I wrongly calculated data size which I wanted to save and I exceeded LFS_FILE_MAX.

Unfortunately I do not have the code which caused it any longer. Requirements in the project changed and I do not have to save so much data any longer.

geky, 2025-02-06:
Ah, if you passed a huge size to `lfs_file_write` that would also cause this. This is easy to do if an earlier function returned a negative error code that went unchecked.

It would have to be > ~2GiB though, since that's the signed 32-bit limit.

But let us know if you run into this again.

BlazejKrysztofiak, 2025-02-06:
Okay, thank you 👍