littlefs-project/littlefs #1067: Does LittleFS put multiple files in a single block? (and a question about append behavior)

vladkorotnev, 2025-02-03:
I'm developing a music player, which would need to cache metadata locally, and for the cache I've decided to use LittleFS.

As an experiment, I've tried caching metadata of 36 albums and monitoring the remaining byte count using the following code:

`ESP_LOGI(LOG_TAG, "Remaining FS size = %i", LittleFS.totalBytes() - LittleFS.usedBytes());`

(of which each LittleFS library call just returns `efs->cfg.block_size * lfs_fs_size(efs->fs);` and `efs->cfg.block_size * efs->cfg.block_count; ` respectively)

However, the results were somewhat strange:

<details>
  <summary>Results table</summary>

CD count | Free Bytes | delta Bytes
-- | -- | --
0 | 839680 |  
1 | 835584 | 4096
2 | 835584 | 0
3 | 831488 | 4096
4 | 827392 | 4096
5 | 823296 | 4096
6 | 823296 | 0
7 | 819200 | 4096
8 | 819200 | 0
9 | 819200 | 0
10 | 819200 | 0
11 | 806912 | 12288
12 | 806912 | 0
13 | 806912 | 0
14 | 806912 | 0
15 | 798720 | 8192
16 | 798720 | 0
17 | 794624 | 4096
18 | 794624 | 0
19 | 794624 | 0
20 | 794624 | 0
21 | 786432 | 8192
22 | 782336 | 4096
23 | 778240 | 4096
24 | 774144 | 4096
25 | 774144 | 0
26 | 774144 | 0
27 | 770048 | 4096
28 | 765952 | 4096
29 | 761856 | 4096
30 | 757760 | 4096
31 | 753664 | 4096
32 | 741376 | 12288
33 | 737280 | 4096
34 | 733184 | 4096
35 | 729088 | 4096
36 | 729088 | 0

</details>

So, on average, the change is +3072 bytes per CD, which doesn't line up with a 4K block size, nor really with a 256b page size either.

Does that mean that creating a new file does not necessarily allocate a new block if it can fit in an existing one? The LittleFS version in my framework is 1.4.x, so definitely before inline files were added.

Additionally, I plan to index the cache by keeping an extra text file with just the IDs of the albums, one on each line. When the line count reaches e.g. 500, the older lines get deleted, and newer ones get appended. Would that cause a performance impact on the filesystem, or prevent wear leveling from correctly operating?

vladkorotnev, 2025-02-04:
Looks like this is indeed the version of the wrapper in ESP-IDF that is linked into the Arduino-ESP32 toolchain:

```
#define ESP_LITTLEFS_VERSION_NUMBER "1.14.1"
#define ESP_LITTLEFS_VERSION_MAJOR 1
#define ESP_LITTLEFS_VERSION_MINOR 14
#define ESP_LITTLEFS_VERSION_PATCH 1
```

Which traces back to https://github.com/joltwallet/esp_littlefs/commit/41873c20fb5cdbcf28d7d6cc04e4bcb4a1305317

Which in turn references https://github.com/littlefs-project/littlefs/commit/f53a0cc961a8acac85f868b431d2f3e58e447ba3 as a submodule, at which point in time: 

https://github.com/littlefs-project/littlefs/blob/f53a0cc961a8acac85f868b431d2f3e58e447ba3/lfs.h#L21-L33

So sorry for the confusion, the version of LFS is indeed 2.9 and not 1.14.1.

I don't see any specific flag enabling inline files, so then I presume they "just work".

The write is done as follows:

https://github.com/vladkorotnev/cd-player/blob/cbe91d8f445f29b23f309448af0dc056296dcd3d/firm/lib/espercdp/src/metadata/caching.cpp#L61-L97

So the block counts are indeed fetched after the file is closed. (I am assuming the std::ofstream close maps directly to the POSIX fclose, which then is a direct mapping to `lfs_file_close` plus some error handling in the wrapper in question)

-----

Given the above, I think it might be that the FS is operating indeed at 4K blocks, and using inline files for small enough cache entries. The huge writes for some CDs might be due to long CDs (20+ tracks) with both per-track artist and track titles containing a lot of unicode characters (2+ byte ones). Adding compression might help alleviate some of that. 

The question then is, would using inline files in such fashion potentially wear out the blocks allocated to the file system directory? From reading the design docs, it seemed like storing everything in a subdirectory would reduce the number of writes to the root block and superblock, but then it might wear out the blocks belonging to those subdirectories. 

Would it make sense, either in terms of performance or write leveling, to "index" the files by creating folders using prefixes of the file name? E.g. rather than storing a cache entry under `/littlefs/cddb/OtoHnE2B.Ng9zdA0RdhB8Dvhu6g-`, do something like `/littlefs/cddb/O/OtoHnE2B.Ng9zdA0RdhB8Dvhu6g-`. Granted, given the cryptographic+base64 nature of the CD IDs, it is very possible that a collection of CDs might yield a few hundred directories with a single file in them — so maybe a hybrid of the former and the latter, depending on whether there is one or more entries starting with the same character?

-----

> Though there is also no API for "deleting" from the beginning of a file, so usually to implement logs like this, you use two files (or some other number) and delete the oldest file when needed.

In this context it's not a log per se, but a list of CD IDs (files) that exist in the order of creation. So the algorithm would go line by line through a C++ fstream until it finds a match, and then copy the rest of the file until the start position of the match, and finally put the match at the end of the file. If there is no match, then it just puts the ID at the end of the file. If there is more than a certain number of files in the list, it would delete the one at the very first line of the file, and do the same copy-to-top dance for that.

For the maximum efficiency of the multi-file approach you mentioned, it would need to be that a single "index block" (single log file) grows as big as 4K to take up a single FS block. Then once the list overflows into a new file, the old one is deleted along with the CDs listed in them. However, at 4K that is a few hundred entries, which might already be too big for the rest of the filesystem to contain data for. 

So, probably this algorithm is not viable or efficient at all and not worth considering, especially given the random writes nuance you've mentioned.

-----

Now, I have noticed the framework I'm using has `CONFIG_LITTLEFS_USE_MTIME` enabled, so there is an mtime attribute available on the file system. (And I'd really not make my own LFS wrapper to use extended attributes properly!)

So the new idea is as follows:

1. When caching a new CD, if NTP time is available (it probably is, otherwise where are we caching from) — set MTIME to current time
2. When a cache hit occurs, and NTP time is available, set MTIME to current time again to mark the cache entry as "recently used"
3. When caching a new CD and the file count is more than some constant, or there is no space on the partition anymore — find the oldest file in the directory and delete, repeat until problem is cleared.

Once again, this raises a question of how catastrophic would this be in terms of wear leveling? Let's say I've run my whole collection of 1000 entries through the cache and now just read out 2-3 entries per day, while the file system has only 1 block remaining — the directory block would be tossed around that one and the one it was previously in, and those two will wear out pretty fast! (well, after about 10k times per block, and I don't think a person can listen to a few CDs 10k times in their entire life, but still)

Of course there is always an option to not touch MTIME and if it yeets something actually frequently accessed... just say it's a cache not a data store 😜 but if possible i'd rather have unused entries die out and used ones stay in memory

geky, 2025-02-06:
> The question then is, **would using inline files in such fashion potentially wear out the blocks allocated to the file system directory?** From reading the design docs, it seemed like storing [...]

I wouldn't worry too much about wearing out the root directory. If the root gets too many writes, littlefs automatically increases the superblock chain, and after one "expansion" this chain in indistinguishable from a subdirectory. More info here: https://github.com/littlefs-project/littlefs/issues/1020#issuecomment-2375435331

> **Would it make sense, either in terms of performance or write leveling, to "index" the files by creating folders using prefixes of the file name?** E.g. rather than storing a cache entry under [...]

Ah yes. This is an area of ongoing work, but at the moment littlefs's directory lookup is $O(n)$, which means breaking extremely large directories up into subdirectories can significantly improve performance.

Eventually, when the B-tree work is stabilized, this shouldn't have an observable difference, but that doesn't exactly help you right now.

Fun fact, you may already be aware of this, but this is actually what git does internally for similar reasons (`ls .git/objects`).

> Granted, given the cryptographic+base64 nature of the CD IDs, it is very possible that a collection of CDs might yield a few hundred directories with a single file in them — so maybe a hybrid of the [...]

If the CD IDs are _truely_ cryptographic, they should end up evenly distributed for every possible prefix of the ID.

That being said, I'm not that familiar with CD IDs, and there's been a lot of flawed "cryptographic" operations in the past :)

> In this context it's not a log per se, but a list of CD IDs (files) that exist in the order of creation. So the algorithm would go line by line through a C++ fstream until it finds a match, and then [...]

Ah, in this case littlefs should perform no worse than any other filesystem, but that does involve writing out the entire file every update. Which isn't the end of the world if this list is significantly smaller than the actually contents of the CDs.

Keeping track of the order-of-magnitude of everything may be useful. You really shouldn't care about kilobytes of metadata if the CDs are megabytes large.

> When caching a new CD and the file count is more than some constant, or there is no space on the partition anymore — find the oldest file in the directory and delete, repeat until problem is cleared.

This sounds like a good plan.

Something worth thinking about is what would the user expect, and if I was a user loading up a song I hadn't listened to in a long time, I wouldn't be surprised if it had to take a bit longer to download.

> **Once again, this raises a question of how catastrophic would this be in terms of wear leveling?** Let's say I've run my whole collection of 1000 entries through the cache and now just read out 2-3 [...]

Good eye! This is one of the main limitations of "dynamic wear-leveling". Blocks containing static data do not contribute to wear-leveling.

LittleFS only implements the weaker form of wear-leveling, "dynamic wear-leveling". This still gives you the property that increasing the disk size increases the lifetime of the device, but it doesn't mean the device's erase cycles are used optimally.

You could put LittleFS on the [Dhara FTL](https://github.com/dlbeer/dhara) to get "static wear-leveling", but this comes with its own set of tradeoffs.

vladkorotnev, 2025-02-20:
Hmm, so I tried implementing my cache with compression now, hoping that all files will mostly fit in the inline space. I also use the first letter of the hash as a directory name. However things became much worse, and now each save eats up 8192 bytes of FS space even for files that are 256-512 bytes.

(the logged size is mostly accurate, there is also a roughly 18 byte header in the file)

```
[   593][I][main.cpp:91] setup(): [FS] Free FS size = 499712
...
[ 23878][I][caching.cpp:213] save_to_cache(): [CDCache] Size before compression = 290
[ 23887][I][caching.cpp:245] save_to_cache(): [CDCache] Size after compression = 214
[ 24104][I][caching.cpp:280] save_to_cache(): [CDCache] Remaining FS capacity after cache = 491520
...
[647463][I][caching.cpp:213] save_to_cache(): [CDCache] Size before compression = 580
[647472][I][caching.cpp:245] save_to_cache(): [CDCache] Size after compression = 393
[647562][I][caching.cpp:280] save_to_cache(): [CDCache] Remaining FS capacity after cache = 483328
...
```

Not sure if it could be related to the fact that now I have several KB of fonts on the same partition, or maybe due to the subfolders? If a subfolder reserves a whole sector, maybe it's better to not use the folder prefix and just store everything in one folder?

geky, 2025-03-13:
> If a subfolder reserves a whole sector, maybe it's better to not use the folder prefix and just store everything in one folder?

Ah you're right, I forgot about this cost.

It doesn't help that my head is split between the current implementation and a fork with ongoing work that doesn't have this issue... But said fork also includes the B-tree backed dirs that make the prefix directories unnecessary sooo...

---

Sorry about the late response, good eye on noticing the block-per-subfolder issue.

It sounds like you could have a threshold for switching between modes. Use a single folder for <1000 CDs and the switch to subfolders >1000 CDs.

I think you're seeing a significant overhead per-file because the ratio of directories to files is high, but as the filesystem fills up, I would expect the distribution of files across directories to end up similar to the distribution of files across multiple metadata blocks in a single directory.

Though as you note this may be overkill if you don't expect users to have problematic numbers of CDs.

That being said, the harsh lesson of filesystems, storage, and computers in general, is that users always eventually try to shove in more data than the thing was designed for :)

vladkorotnev, 2025-03-13:
Thanks for the confirmation! It's good enough for now, and I don't think anyone would have a collection that would actually yield a uniform distribution across folders before running out of the measly 836KB of space remaining. 

Your help is very much appreciated. I'll close this since this isn't really an issue and we've figured out the details.

-----

BTW, if you're curious to check out the project, I've made a small demo video :-) https://youtu.be/To1woYCxPkw

closed 2025-03-13 by vladkorotnev